Lienol/openwrt #311: dcoker无法ping通

lexin8, 2020-05-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
容器获得IP172.17.0.3,电脑网页可以打开可是PING不通，导致APP无法连接打开服务。
A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

zhiapp-com, 2020-05-22:
确实这个版本用docker运行。外网不通，不知道如何解决。

zhiapp-com, 2020-05-22:
/etc/resovl.conf 
nameserver 127.0.0.11 
为什么是这个呢，不是127.0.0.1

lexin8, 2020-06-06:
macvlan自定义网络无法PING通外网，路由IP都PING不通。liunx系统下安装docker测试是可以访问外网正常的。

lexin8, 2020-06-20:
dev-19.07、dev-master都试过了，同样问题。
[image]
[image]
[image]

lexin8, 2020-06-20:
> 更新最新源码 用最新的luci试试
dev-19.07、dev-master都试过了，同样问题。

Lienol, 2020-06-20:
可能是op问题吧，我没用op的docker

Lienol, 2020-06-20:
[image]
[image]

测试正常

lexin8, 2020-06-21:
> [image]
> [image]
> 
> 测试正常

你这个是bridge/host模式吧？
macvlan自定义网络（指定路由局域网IP）

lexin8, 2020-06-21:
> [image]
> [image]
> 
> 测试正常

> > [image]
> > [image]
> > 测试正常
> 
> 你这个是bridge/host模式吧？
> macvlan自定义网络（指定路由局域网IP）
换成路由网段试试呢 192.168.1.1

lexin8, 2020-06-21:
[image]
这是Ubuntu里安装docker运行测试是可以通的。

closed 2020-07-04 by lexin8