Lienol/openwrt #313: luci-app-ssr-python-pro-server有人备份么？

xxp2277, 2020-05-23:
众所周知的原因luci-app-ssr-python-pro-server被删除了，但是我还是想用，有人将这个项目备份么？

Kisoul, 2020-05-26:
有备份但是编译了启动不了不知道为什么

Lienol, 2020-05-27:
检查脚本权限

closed 2020-05-27 by Lienol

czpwpq, 2020-09-02:
make package/lean/luci-app-ssr-python-pro-server/{clean,compile} V=s
WARNING: Makefile 'package/feeds/laohu/luci-app-ssr-python-pro-server/Makefile' has a dependency on 'python', which does not exist
make[1]: Entering directory '/home/czpwpq/lede'
make[1]: *** No rule to make target 'package/lean/luci-app-ssr-python-pro-server/clean'.  Stop.
make[1]: Leaving directory '/home/czpwpq/lede'
make: *** [/home/czpwpq/lede/include/toplevel.mk:224: package/lean/luci-app-ssr-python-pro-server/clean] Error 2

就这个包过不了，哎，不知道原因