Lienol/openwrt #316: 编译19.07版x86报错。

jlzsss, 2020-05-28:
报错信息如下：
2020-05-27T16:15:13.4982848Z  * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-fs-ksmbd:
2020-05-27T16:15:13.4983128Z Configuring strongswan-mod-updown.
2020-05-27T16:15:13.4984609Z  * 	kmod-crypto-arc4
2020-05-27T16:15:13.4984912Z Configuring luci-app-mjpg-streamer.
2020-05-27T16:15:13.4987671Z  * opkg_install_cmd: Cannot install package kmod-fs-ksmbd.
2020-05-27T16:15:13.4987957Z Configuring luci-i18n-mjpg-streamer-zh-cn.
2020-05-27T16:15:13.4989466Z  * resolve_conffiles: Existing conffile /home/runner/work/OpenWRT-X86_64/OpenWRT-X86_64/openwrt/build_dir/target-x86_64_musl/root-x86/etc/ppp/options is different from the conffile in the new package. The new conffile will be placed at /home/runner/work/OpenWRT-X86_64/OpenWRT-X86_64/openwrt/build_dir/target-x86_64_musl/root-x86/etc/ppp/options-opkg.
2020-05-27T16:15:13.4989798Z Configuring ath10k-firmware-qca9888.
2020-05-27T16:15:13.4990725Z make[2]: *** [package/install] Error 255

make menuconfig时提示如下信息：
WARNING: Makefile 'package/lean/ksmbd-tools/ksmbd/Makefile' has a dependency on 'kmod-crypto-arc4', which does not exist
tmp/.config-package.in:117202:error: recursive dependency detected!

danety, 2020-05-28:
我也遇到同样的问题

hong0980, 2020-05-29:
我是`sed -i '/+kmod-crypto-arc4/d' package/lean/ksmbd/Makefile
`删除这个依赖。

jlzsss, 2020-05-30:
最新版编译成功。

Lienol, 2020-05-30:
Fix

closed 2020-05-30 by Lienol