Lienol/openwrt #330: 开启bbr后遇到的现象，不知道算不算问题

roycho93, 2020-06-23:
不知道算不算bug吧，只是遇到一个现象想问问大神能不能解答疑惑

 家里宽带是100M上行，平时nas中的资料经常会备份到谷歌网盘中，所以备份的时候就会占满上行带宽，之前没发现什么异样，昨天换了一张万兆网卡（x540-t2），因为光猫lan口是千兆的，所以网卡还是工作在千兆模式下，发现上行占满的时候家里上网竟然有卡顿现象出现，找了半天原因，发现一些端倪。
  nas在上传资料到谷歌云盘的过程中始终是13M左右，最高15M出头，不管用哪张网卡都一样，因为上行带宽就限制在这里了，那为什么用x540的时候上行占满就会上网卡顿呢，于是我看了下netdata中显示的上行数据，显示上行远远不止13M，13M*8=108m，按理说应该显示108左右的上行，可是netdata里用千兆网卡时显示120m的上行，用了x540竟然有140m的上行，但是实际nas在上传时候速度都差不多，多出来的流量我想到了bbr，关闭了后实际nas传输速度有所下降 大概到了11M左右，netdata里显示也降到大概100左右的样子了，于是我想应该就是开启bbr后两张网卡一个是120m一个是140m上行导致的，因为平时speedtest测速上行也就122左右，140应该是本地网卡产生的数据量？但是上行带宽限制了所以拥堵造成了卡顿，于是乎就在SQM Qos这个软件限制了wan口上行速率到120，这下x540也不卡顿了，nas的实际上传速度也没有明显的变化，在13-15M左右。
  就像知道为什么不同的两张网卡会造成bbr开启后netdata这边上行速度有20m左右差异的问题，比较好奇，而且用x540时限制wan口上传120m后实际nas速度也没有明显变化。

roycho93, 2020-06-23:
因为问题出在 bbr 上，后来我找到软件包中找到 kmod-tcp-bbr，想着换个版本试一下，原先版本是 4.14.181-1,应该已经是最新版本了，换到 4.14.161-1，诡异的事来了，nas 中上传速度还是 13M 左右，netdata 中 wan 口直接飙升到 250m 的上行，再换回 4.14.181-1 还是一样，现在折腾了半天都没用

Lienol, 2020-06-25:
不懂

closed 2020-06-25 by Lienol