Lienol/openwrt #331: 编译dev-19.07-5.4版x86-64报错。

jlzsss, 2020-06-23:
报错内容如下：
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'bnx2x-firmware', which does not exist

jlzsss, 2020-06-25:
第一个问题解决了，可是我又编译一次，还是报错。
报错如下：
2020-06-25T05:17:16.0839168Z drivers/net/phy/swconfig.c:1004:4: error: 'struct genl_ops' has no member named 'policy'
2020-06-25T05:17:16.0839953Z    .policy = switch_policy,
2020-06-25T05:17:16.0840323Z     ^~~~~~
2020-06-25T05:17:16.0841067Z drivers/net/phy/swconfig.c:1004:13: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
2020-06-25T05:17:16.0842261Z    .policy = switch_policy,
2020-06-25T05:17:16.0842763Z              ^~~~~~~~~~~~~
2020-06-25T05:17:16.0843283Z drivers/net/phy/swconfig.c:1004:13: note: (near initialization for 'swconfig_ops[0].start')
2020-06-25T05:17:16.0843869Z drivers/net/phy/swconfig.c:1009:4: error: 'struct genl_ops' has no member named 'policy'
2020-06-25T05:17:16.0844231Z    .policy = switch_policy,
2020-06-25T05:17:16.0844522Z     ^~~~~~
2020-06-25T05:17:16.0845083Z drivers/net/phy/swconfig.c:1009:13: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
2020-06-25T05:17:16.0845448Z    .policy = switch_policy,
2020-06-25T05:17:16.0845743Z              ^~~~~~~~~~~~~
2020-06-25T05:17:16.0846241Z drivers/net/phy/swconfig.c:1009:13: note: (near initialization for 'swconfig_ops[1].start')
2020-06-25T05:17:16.0846981Z drivers/net/phy/swconfig.c:1014:4: error: 'struct genl_ops' has no member named 'policy'
2020-06-25T05:17:16.0847343Z    .policy = switch_policy,
2020-06-25T05:17:16.0847620Z     ^~~~~~
2020-06-25T05:17:16.0848158Z drivers/net/phy/swconfig.c:1014:13: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
2020-06-25T05:17:16.0848509Z    .policy = switch_policy,
2020-06-25T05:17:16.0848790Z              ^~~~~~~~~~~~~
2020-06-25T05:17:16.0849262Z drivers/net/phy/swconfig.c:1014:13: note: (near initialization for 'swconfig_ops[2].start')
2020-06-25T05:17:16.0849811Z drivers/net/phy/swconfig.c:1019:4: error: 'struct genl_ops' has no member named 'policy'
2020-06-25T05:17:16.0850158Z    .policy = switch_policy,
2020-06-25T05:17:16.0850443Z     ^~~~~~
2020-06-25T05:17:16.0850958Z drivers/net/phy/swconfig.c:1019:13: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
2020-06-25T05:17:16.0851550Z    .policy = switch_policy,
2020-06-25T05:17:16.0851843Z              ^~~~~~~~~~~~~
2020-06-25T05:17:16.0852342Z drivers/net/phy/swconfig.c:1019:13: note: (near initialization for 'swconfig_ops[3].start')
2020-06-25T05:17:16.0853158Z drivers/net/phy/swconfig.c:1024:4: error: 'struct genl_ops' has no member named 'policy'
2020-06-25T05:17:16.0853527Z    .policy = switch_policy,
2020-06-25T05:17:16.0853810Z     ^~~~~~
2020-06-25T05:17:16.0854514Z drivers/net/phy/swconfig.c:1024:13: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
2020-06-25T05:17:16.0854865Z    .policy = switch_policy,
2020-06-25T05:17:16.0855119Z              ^~~~~~~~~~~~~
2020-06-25T05:17:16.0855607Z drivers/net/phy/swconfig.c:1024:13: note: (near initialization for 'swconfig_ops[4].start')
2020-06-25T05:17:16.0856209Z drivers/net/phy/swconfig.c:1029:4: error: 'struct genl_ops' has no member named 'policy'
2020-06-25T05:17:16.0856567Z    .policy = switch_policy,
2020-06-25T05:17:16.0856848Z     ^~~~~~
2020-06-25T05:17:16.0857367Z drivers/net/phy/swconfig.c:1029:13: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
2020-06-25T05:17:16.0857732Z    .policy = switch_policy,
2020-06-25T05:17:16.0858023Z              ^~~~~~~~~~~~~
2020-06-25T05:17:16.0858708Z drivers/net/phy/swconfig.c:1029:13: note: (near initialization for 'swconfig_ops[5].start')
2020-06-25T05:17:16.0859339Z drivers/net/phy/swconfig.c:1035:4: error: 'struct genl_ops' has no member named 'policy'
2020-06-25T05:17:16.0859696Z    .policy = switch_policy,
2020-06-25T05:17:16.0859986Z     ^~~~~~
2020-06-25T05:17:16.0860514Z drivers/net/phy/swconfig.c:1035:13: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
2020-06-25T05:17:16.0860870Z    .policy = switch_policy,
2020-06-25T05:17:16.0861160Z              ^~~~~~~~~~~~~
2020-06-25T05:17:16.0861689Z drivers/net/phy/swconfig.c:1035:13: note: (near initialization for 'swconfig_ops[6].start')
2020-06-25T05:17:16.0862250Z drivers/net/phy/swconfig.c:1041:4: error: 'struct genl_ops' has no member named 'policy'
2020-06-25T05:17:16.0862601Z    .policy = switch_policy,
2020-06-25T05:17:16.0863056Z     ^~~~~~
2020-06-25T05:17:16.0863627Z drivers/net/phy/swconfig.c:1041:13: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
2020-06-25T05:17:16.0864016Z    .policy = switch_policy,
2020-06-25T05:17:16.0864851Z              ^~~~~~~~~~~~~
2020-06-25T05:17:16.0865561Z drivers/net/phy/swconfig.c:1041:13: note: (near initialization for 'swconfig_ops[7].start')
2020-06-25T05:17:16.0866028Z drivers/net/phy/swconfig.c:1047:4: error: 'struct genl_ops' has no member named 'policy'
2020-06-25T05:17:16.0866382Z    .policy = switch_policy,
2020-06-25T05:17:16.0866622Z     ^~~~~~
2020-06-25T05:17:16.0867112Z drivers/net/phy/swconfig.c:1047:13: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
2020-06-25T05:17:16.0867411Z    .policy = switch_policy,
2020-06-25T05:17:16.0867647Z              ^~~~~~~~~~~~~
2020-06-25T05:17:16.0868209Z drivers/net/phy/swconfig.c:1047:13: note: (near initialization for 'swconfig_ops[8].start')
[... cut ...]
2020-06-25T05:17:16.1162134Z make: *** [world] Error 2
2020-06-25T05:17:16.1162375Z make[2]: Leaving directory '/workdir/openwrt'
2020-06-25T05:17:16.1162662Z target/Makefile:18: recipe for target '/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.target_compile' failed
2020-06-25T05:17:16.1162894Z make[1]: Leaving directory '/workdir/openwrt'
2020-06-25T05:17:16.1163135Z /workdir/openwrt/include/toplevel.mk:225: recipe for target 'world' failed
2020-06-25T05:17:16.1164987Z ##[error]Process completed with exit code 2.

Lienol, 2020-06-25:
未完全移植好
用dev-master分支

closed 2020-06-30 by Lienol