Lienol/openwrt #333: 编译dev master出错，请大神帮忙看一下，感谢！

trombonist852, 2020-06-24:
路由器是x86软路由
第一个问题是make menuconfig之后会报两个警告
WARNING: Makefile 'package/feeds/luci/luci-app-e2guardian/Makefile' has a dependency on 'e2guardian', which does not exist
WARNING: Makefile 'package/lean/qt5/Makefile' has a dependency on 'libdouble-conversion', which does not exist

第二个问题是编译到最后会跳如下错误不知道怎么解决，望大神指点一二，感谢！

* check_data_file_clashes: Package libgd-full wants to install file /home/tr0mb0n1st/op-dev-master/build_dir/target-x86_64_musl/root-x86/usr/lib/libgd.so
	But that file is already provided by package  * libgd
 * check_data_file_clashes: Package libgd-full wants to install file /home/tr0mb0n1st/op-dev-master/build_dir/target-x86_64_musl/root-x86/usr/lib/libgd.so.3
	But that file is already provided by package  * libgd
 * check_data_file_clashes: Package libgd-full wants to install file /home/tr0mb0n1st/op-dev-master/build_dir/target-x86_64_musl/root-x86/usr/lib/libgd.so.3.0.8
	But that file is already provided by package  * libgd
 * opkg_install_cmd: Cannot install package libgd-full.
 * check_data_file_clashes: Package libgd-full wants to install file /home/tr0mb0n1st/op-dev-master/build_dir/target-x86_64_musl/root-x86/usr/lib/libgd.so
	But that file is already provided by package  * libgd
 * check_data_file_clashes: Package libgd-full wants to install file /home/tr0mb0n1st/op-dev-master/build_dir/target-x86_64_musl/root-x86/usr/lib/libgd.so.3
	But that file is already provided by package  * libgd
 * check_data_file_clashes: Package libgd-full wants to install file /home/tr0mb0n1st/op-dev-master/build_dir/target-x86_64_musl/root-x86/usr/lib/libgd.so.3.0.8
	But that file is already provided by package  * libgd
 * opkg_install_cmd: Cannot install package php7-mod-gd.
 * resolve_conffiles: Existing conffile /home/tr0mb0n1st/op-dev-master/build_dir/target-x86_64_musl/root-x86/etc/ppp/options is different from the conffile in the new package. The new conffile will be placed at /home/tr0mb0n1st/op-dev-master/build_dir/target-x86_64_musl/root-x86/etc/ppp/options-opkg.
make[2]: *** [package/Makefile:68: package/install] Error 255
make[2]: Leaving directory '/home/tr0mb0n1st/op-dev-master'
make[1]: *** [package/Makefile:108: /home/tr0mb0n1st/op-dev-master/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/tr0mb0n1st/op-dev-master'
make: *** [/home/tr0mb0n1st/op-dev-master/include/toplevel.mk:224：world] 错误 2

adameng, 2020-06-25:
ARM编译同样出现第一个问题，dev-19.07和dev-19.07-5.4没问题，望大神指点

Lienol, 2020-06-25:
libgd选了吗

trombonist852, 2020-06-25:
选啦，但是libgd下面一项是libgd-full，我尝试把full这一项取消但是取消不了…按n没用，前面的*取消不了，貌似是强制编译进去的

Lienol, 2020-06-25:
那看看能不能把libgd取消掉，勾选libgd-full

trombonist852, 2020-06-25:
尝试过啦，这两个都不能取消，按n都没用*一直都在…

Lienol, 2020-06-25:
两个不能同时选，自己处理下吧。
你应该是选了一些依赖了libgd的包

closed 2020-06-25 by Lienol

Lienol, 2020-06-25:
优化了一下，更新feeds，然后make menuconfig菜单里面取消掉libgd-full编译即可

trombonist852, 2020-06-26:
我改用19.07-5.4的分支了😂，目前还没有报错，谢谢大神啦！