Lienol/openwrt #338: 编译19.07-5.4错误，望大神帮忙分析以下，感谢！

trombonist852, 2020-06-26:
A：大神，我在用-5.4编译的时候会跳出这个错误，能否帮忙分析一下，感激不尽！

### (3) 详细日志

A：/usr/bin/ld: warning: libc.so, needed by /home/tr0mb0n1st/op5.4/staging_dir/target-x86_64_musl/usr/lib/libelf.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: /home/tr0mb0n1st/op5.4/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.48/tools/objtool/objtool: hidden symbol `fstat' in /usr/lib/x86_64-linux-gnu/libc_nonshared.a(fstat.oS) is referenced by DSO
/usr/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
make[6]: *** [Makefile:50: /home/tr0mb0n1st/op5.4/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.48/tools/objtool/objtool] Error 1
make[5]: *** [Makefile:67: objtool] Error 2
make[4]: *** [Makefile:1811: tools/objtool] Error 2
make[4]: Leaving directory '/home/tr0mb0n1st/op5.4/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.48'
make[3]: *** [Makefile:73: /home/tr0mb0n1st/op5.4/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.1.3/.built] Error 2
make[3]: Leaving directory '/home/tr0mb0n1st/op5.4/feeds/packages/kernel/ksmbd'
time: package/feeds/packages/ksmbd/compile#8.09#0.91#8.99
make[2]: *** [package/Makefile:113: package/feeds/packages/ksmbd/compile] Error 2
make[2]: Leaving directory '/home/tr0mb0n1st/op5.4'
make[1]: *** [package/Makefile:107: /home/tr0mb0n1st/op5.4/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/tr0mb0n1st/op5.4'
make: *** [/home/tr0mb0n1st/op5.4/include/toplevel.mk:227：world] 错误 2

Lienol, 2020-06-27:
重拉源码，再试试

trombonist852, 2020-06-28:
好的，我试试，谢谢大神！

closed 2020-06-30 by Lienol