Lienol/openwrt #345: 编译dev-19.07版x86-64报错，请大神帮忙看看。

jlzsss, 2020-07-02:
报错如下：
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/lang/python/python-six'
2020-07-01T20:33:55.4955132Z time: package/feeds/packages/python-six/python3/compile#0.20#0.05#0.24
2020-07-01T20:33:55.5341540Z make[3]: Entering directory '/workdir/openwrt/feeds/packages/lang/python/python-dateutil'
2020-07-01T20:33:55.6707459Z rm -f /workdir/openwrt/build_dir/target-x86_64_musl/python3-dateutil-2.8.0/.built
2020-07-01T20:33:55.6736935Z touch /workdir/openwrt/build_dir/target-x86_64_musl/python3-dateutil-2.8.0/.built_check
2020-07-01T20:33:55.6760167Z install -d -m0755 /workdir/openwrt/build_dir/target-x86_64_musl/python3-dateutil-2.8.0/ipkg-install//usr/lib/python3.7/site-packages
2020-07-01T20:33:55.6791216Z if [ "" == "HOST" ]; then export PYTHONPATH="/workdir/openwrt/staging_dir/hostpkg/lib/python3.7:/workdir/openwrt/staging_dir/hostpkg/lib/python3.7/site-packages"; export PYTHONDONTWRITEBYTECODE=0; else export PYTHONPATH="/workdir/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.7:/workdir/openwrt/staging_dir/target-x86_64_musl//usr/lib/python3.7/site-packages:/workdir/openwrt/build_dir/target-x86_64_musl/python3-dateutil-2.8.0/ipkg-install//usr/lib/python3.7/site-packages"; export PYTHONDONTWRITEBYTECODE=1; export _python_sysroot="/workdir/openwrt/staging_dir/target-x86_64_musl"; export _python_prefix="/usr"; export _python_exec_prefix="/usr"; fi; export PYTHONOPTIMIZE="";   cd /workdir/openwrt/build_dir/target-x86_64_musl/python3-dateutil-2.8.0/; CC="x86_64-openwrt-linux-musl-gcc" CCSHARED="x86_64-openwrt-linux-musl-gcc -fpic" CXX="x86_64-openwrt-linux-musl-g++" LD="x86_64-openwrt-linux-musl-gcc" LDSHARED="x86_64-openwrt-linux-musl-gcc -shared" CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/workdir/openwrt/build_dir/target-x86_64_musl/python3-dateutil-2.8.0:python3-dateutil-2.8.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CPPFLAGS="-I/workdir/openwrt/staging_dir/target-x86_64_musl/usr/include -I/workdir/openwrt/staging_dir/target-x86_64_musl/include -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include/fortify -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include -I/workdir/openwrt/staging_dir/target-x86_64_musl/usr/include/python3.7" LDFLAGS="-L/workdir/openwrt/staging_dir/target-x86_64_musl/usr/lib -L/workdir/openwrt/staging_dir/target-x86_64_musl/lib -L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/usr/lib -L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/lib -znow -zrelro -lpython3.7" _PYTHON_HOST_PLATFORM=linux2 __PYVENV_LAUNCHER__="/usr/bin/python3.7"      /workdir/openwrt/staging_dir/hostpkg/bin/python3.7   ./setup.py   install --prefix="/usr" --root="/workdir/openwrt/build_dir/target-x86_64_musl/python3-dateutil-2.8.0/ipkg-install" --single-version-externally-managed ;
2020-07-01T20:33:56.7573895Z WARNING: The wheel package is not available.
2020-07-01T20:34:04.0706279Z ERROR: Exception:
2020-07-01T20:34:04.0706947Z Traceback (most recent call last):
2020-07-01T20:34:04.0708043Z   File "/workdir/openwrt/staging_dir/hostpkg/lib/python3.7/site-packages/pip/_internal/cli/base_command.py", line 188, in _main
2020-07-01T20:34:04.0708228Z     status = self.run(options, args)
2020-07-01T20:34:04.0708653Z   File "/workdir/openwrt/staging_dir/hostpkg/lib/python3.7/site-packages/pip/_internal/cli/req_command.py", line 185, in wrapper
2020-07-01T20:34:04.0708884Z     return func(self, options, args)
2020-07-01T20:34:04.0709290Z   File "/workdir/openwrt/staging_dir/hostpkg/lib/python3.7/site-packages/pip/_internal/commands/wheel.py", line 160, in run
2020-07-01T20:34:04.0709449Z     reqs, check_supported_wheels=True
2020-07-01T20:34:04.0709863Z   File "/workdir/openwrt/staging_dir/hostpkg/lib/python3.7/site-packages/pip/_internal/resolution/legacy/resolver.py", line 179, in resolve
2020-07-01T20:34:04.0710157Z     discovered_reqs.extend(self._resolve_one(requirement_set, req))
2020-07-01T20:34:04.0710602Z   File "/workdir/openwrt/staging_dir/hostpkg/lib/python3.7/site-packages/pip/_internal/resolution/legacy/resolver.py", line 362, in _resolve_one
2020-07-01T20:34:04.0710772Z     abstract_dist = self._get_abstract_dist_for(req_to_install)
2020-07-01T20:34:04.0711209Z   File "/workdir/openwrt/staging_dir/hostpkg/lib/python3.7/site-packages/pip/_internal/resolution/legacy/resolver.py", line 313, in _get_abstract_dist_for
2020-07-01T20:34:04.0711438Z     self._populate_link(req)
2020-07-01T20:34:04.0711847Z   File "/workdir/openwrt/staging_dir/hostpkg/lib/python3.7/site-packages/pip/_internal/resolution/legacy/resolver.py", line 279, in _populate_link
2020-07-01T20:34:04.0712016Z     req.link = self.finder.find_requirement(req, upgrade)
2020-07-01T20:34:04.0712415Z   File "/workdir/openwrt/staging_dir/hostpkg/lib/python3.7/site-packages/pip/_internal/index/package_finder.py", line 901, in find_requirement
2020-07-01T20:34:04.0712699Z     req.name, specifier=req.specifier, hashes=hashes,
2020-07-01T20:34:04.0713267Z   File "/workdir/openwrt/staging_dir/hostpkg/lib/python3.7/site-packages/pip/_internal/index/package_finder.py", line 883, in find_best_candidate
2020-07-01T20:34:04.0713425Z     candidates = self.find_all_candidates(project_name)
2020-07-01T20:34:04.0713828Z   File "/workdir/openwrt/staging_dir/hostpkg/lib/python3.7/site-packages/pip/_internal/index/package_finder.py", line 828, in find_all_candidates
2020-07-01T20:34:04.0713978Z     project_url, link_evaluator=link_evaluator,
2020-07-01T20:34:04.0714368Z   File "/workdir/openwrt/staging_dir/hostpkg/lib/python3.7/site-packages/pip/_internal/index/package_finder.py", line 801, in process_project_url
2020-07-01T20:34:04.0714518Z     links=page_links,
2020-07-01T20:34:04.0714907Z   File "/workdir/openwrt/staging_dir/hostpkg/lib/python3.7/site-packages/pip/_internal/index/package_finder.py", line 781, in evaluate_links
2020-07-01T20:34:04.0715442Z     candidate = self.get_install_candidate(link_evaluator, link)
2020-07-01T20:34:04.0716007Z   File "/workdir/openwrt/staging_dir/hostpkg/lib/python3.7/site-packages/pip/_internal/index/package_finder.py", line 760, in get_install_candidate
2020-07-01T20:34:04.0716154Z     is_candidate, result = link_evaluator.evaluate_link(link)
2020-07-01T20:34:04.0716554Z   File "/workdir/openwrt/staging_dir/hostpkg/lib/python3.7/site-packages/pip/_internal/index/package_finder.py", line 197, in evaluate_link
2020-07-01T20:34:04.0716944Z     supported_tags = self._target_python.get_tags()
2020-07-01T20:34:04.0717348Z   File "/workdir/openwrt/staging_dir/hostpkg/lib/python3.7/site-packages/pip/_internal/models/target_python.py", line 106, in get_tags
2020-07-01T20:34:04.0717507Z     impl=self.implementation,
2020-07-01T20:34:04.0717915Z   File "/workdir/openwrt/staging_dir/hostpkg/lib/python3.7/site-packages/pip/_internal/utils/compatibility_tags.py", line 150, in get_supported
2020-07-01T20:34:04.0718068Z     platforms=platforms,
2020-07-01T20:34:04.0718484Z   File "/workdir/openwrt/staging_dir/hostpkg/lib/python3.7/site-packages/pip/_vendor/packaging/tags.py", line 239, in cpython_tags
2020-07-01T20:34:04.0718642Z     platforms = list(platforms or _platform_tags())
2020-07-01T20:34:04.0719040Z   File "/workdir/openwrt/staging_dir/hostpkg/lib/python3.7/site-packages/pip/_vendor/packaging/tags.py", line 642, in _linux_platforms
2020-07-01T20:34:04.0719193Z     _, arch = linux.split("_", 1)
2020-07-01T20:34:04.0719310Z ValueError: not enough values to unpack (expected 2, got 1)
2020-07-01T20:34:04.1311005Z Traceback (most recent call last):
2020-07-01T20:34:04.1312002Z   File "/workdir/openwrt/staging_dir/hostpkg/lib/python3.7/site-packages/setuptools/installer.py", line 128, in fetch_build_egg
2020-07-01T20:34:04.1312218Z     subprocess.check_call(cmd)
2020-07-01T20:34:04.1312624Z   File "/workdir/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.7/subprocess.py", line 363, in check_call
2020-07-01T20:34:04.1312781Z     raise CalledProcessError(retcode, cmd)
2020-07-01T20:34:04.1313390Z subprocess.CalledProcessError: Command '['/workdir/openwrt/staging_dir/hostpkg/bin/python3.7', '-m', 'pip', '--disable-pip-version-check', 'wheel', '--no-deps', '-w', '/workdir/openwrt/tmp/tmph3bsdhcp', '--quiet', 'setuptools_scm']' returned non-zero exit status 2.
[... cut ...]
2020-07-01T20:34:04.1725877Z make[2]: Leaving directory '/workdir/openwrt'
2020-07-01T20:34:04.1726902Z package/Makefile:107: recipe for target '/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
2020-07-01T20:34:04.1750086Z make[1]: Leaving directory '/workdir/openwrt'
2020-07-01T20:34:04.1765373Z /workdir/openwrt/include/toplevel.mk:225: recipe for target 'world' failed
2020-07-01T20:34:04.1769789Z make: *** [world] Error 2
2020-07-01T20:34:04.1773507Z ##[error]Process completed with exit code 2.

Lienol, 2020-07-04:
更新一下python3再make clean 试试

jlzsss, 2020-07-04:
更新python3以后，重新编译还是报同样的错误。

closed 2020-07-16 by Lienol