Lienol/openwrt #348: dev-master编译qbittorrent时出错

wjlin2k, 2020-07-05:
dev-master最新源码编译qbittorrent用到的rblibtorrent时出错，前几天Lean的源码也出同样的错误，但现已修复

OpenWrt-libtool: compile:  x86_64-openwrt-linux-musl-g++ -DPACKAGE_NAME=\"libtorrent-rasterbar\" -DPACKAGE_TARNAME=\"libtorrent-rasterbar\" -DPACKAGE_VERSION=\"1.1.13\" "-DPACKAGE_STRING=\"libtorrent-rasterbar 1.1.13\"" -DPACKAGE_BUGREPORT=\"arvid@libtorrent.org\" -DPACKAGE_URL=\"http://www.libtorrent.org\" -DPACKAGE=\"libtorrent-rasterbar\" -DVERSION=\"1.1.13\" -DSTDC_HEADERS=1 -DHAVE_SYS_TYPES_H=1 -DHAVE_SYS_STAT_H=1 -DHAVE_STDLIB_H=1 -DHAVE_STRING_H=1 -DHAVE_MEMORY_H=1 -DHAVE_STRINGS_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_UNISTD_H=1 -DHAVE_DLFCN_H=1 -DLT_OBJDIR=\".libs/\" -DHAVE_PTHREAD=1 "-DHAVE_BOOST=/**/" "-DHAVE_BOOST_SYSTEM=/**/" "-DHAVE_BOOST_CHRONO=/**/" "-DHAVE_BOOST_RANDOM=/**/" -DHAVE_CLOCK_GETTIME=1 -DNDEBUG=1 -DTORRENT_USE_OPENSSL=1 -DHAVE_ICONV=1 -DICONV_CONST= -DBOOST_ASIO_HASH_MAP_BUCKETS=1021 -DBOOST_EXCEPTION_DISABLE=1 -DBOOST_ASIO_ENABLE_CANCELIO=1 -DTORRENT_BUILDING_SHARED=1 -I. -DTORRENT_BUILDING_LIBRARY -I../include -I../ed25519/src -Os -I/home/gly/op-dev-master/staging_dir/target-x86_64_musl/usr/include -I/home/gly/op-dev-master/staging_dir/target-x86_64_musl/usr/include -I/home/gly/op-dev-master/staging_dir/target-x86_64_musl/usr/include -I/home/gly/op-dev-master/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/gly/op-dev-master/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/gly/op-dev-master/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/gly/op-dev-master/build_dir/target-x86_64_musl/rblibtorrent-1.1.13=rblibtorrent-1.1.13 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fPIC -std=gnu++14 -ftemplate-depth=120 -Wno-format-zero-length -fvisibility=hidden -fvisibility-inlines-hidden -MT http_connection.lo -MD -MP -MF .deps/http_connection.Tpo -c http_connection.cpp  -fPIC -DPIC -o .libs/http_connection.o
In file included from ../include/libtorrent/http_connection.hpp:41,
                 from http_connection.cpp:33:
/home/gly/op-dev-master/staging_dir/target-x86_64_musl/usr/include/boost/bind.hpp:41:1: note: #pragma message: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.
 )
 ^
In file included from /home/gly/op-dev-master/staging_dir/target-x86_64_musl/usr/include/boost/asio/write.hpp:1246,
                 from ../include/libtorrent/socket.hpp:57,
                 from ../include/libtorrent/http_connection.hpp:62,
                 from http_connection.cpp:33:
/home/gly/op-dev-master/staging_dir/target-x86_64_musl/usr/include/boost/asio/impl/write.hpp: In instantiation of 'class boost::asio::detail::initiate_async_write_buffer_sequence<libtorrent::socket_type>':
/home/gly/op-dev-master/staging_dir/target-x86_64_musl/usr/include/boost/asio/impl/write.hpp:534:15:   required from 'auto boost::asio::async_write(AsyncWriteStream&, const ConstBufferSequence&, WriteHandler&&, typename std::enable_if<boost::asio::is_const_buffer_sequence<ConstBufferSequence>::value>::type*) [with AsyncWriteStream = libtorrent::socket_type; ConstBufferSequence = boost::asio::mutable_buffers_1; WriteHandler = boost::_bi::bind_t<void, boost::_mfi::mf1<void, libtorrent::http_connection, const boost::system::error_code&>, boost::_bi::list2<boost::_bi::value<boost::shared_ptr<libtorrent::http_connection> >, boost::arg<1> > >; typename std::enable_if<boost::asio::is_const_buffer_sequence<ConstBufferSequence>::value>::type = void]'
http_connection.cpp:278:53:   required from here
/home/gly/op-dev-master/staging_dir/target-x86_64_musl/usr/include/boost/asio/impl/write.hpp:425:54: error: no type named 'executor_type' in 'struct libtorrent::socket_type'
     typedef typename AsyncWriteStream::executor_type executor_type;
                                                      ^~~~~~~~~~~~~
Makefile:937: recipe for target 'http_connection.lo' failed
make[5]: *** [http_connection.lo] Error 1
make[5]: Leaving directory '/home/gly/op-dev-master/build_dir/target-x86_64_musl/rblibtorrent-1.1.13/src'
Makefile:623: recipe for target 'all-recursive' failed
make[4]: *** [all-recursive] Error 1
make[4]: Leaving directory '/home/gly/op-dev-master/build_dir/target-x86_64_musl/rblibtorrent-1.1.13'
Makefile:74: recipe for target '/home/gly/op-dev-master/build_dir/target-x86_64_musl/rblibtorrent-1.1.13/.built' failed
make[3]: *** [/home/gly/op-dev-master/build_dir/target-x86_64_musl/rblibtorrent-1.1.13/.built] Error 2
make[3]: Leaving directory '/home/gly/op-dev-master/package/lean/rblibtorrent'
time: package/lean/rblibtorrent/compile#2.95#0.44#3.34
package/Makefile:111: recipe for target 'package/lean/rblibtorrent/compile' failed
make[2]: *** [package/lean/rblibtorrent/compile] Error 2
make[2]: Leaving directory '/home/gly/op-dev-master'
package/Makefile:107: recipe for target '/home/gly/op-dev-master/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/gly/op-dev-master/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/gly/op-dev-master'
/home/gly/op-dev-master/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

Lienol, 2020-07-05:
更新源码再试试

wjlin2k, 2020-07-05:
还会出错

make[3]: Entering directory '/home/gly/op-dev-master/package/lean/qBittorrent'
rm -f /home/gly/op-dev-master/build_dir/target-x86_64_musl/qBittorrent-release-4.2.3/.configured_*
rm -f /home/gly/op-dev-master/staging_dir/target-x86_64_musl/stamp/.qBittorrent_installed
(cd /home/gly/op-dev-master/build_dir/target-x86_64_musl/qBittorrent-release-4.2.3/./; if [ -x ./configure ]; then find /home/gly/op-dev-master/build_dir/target-x86_64_musl/qBittorrent-release-4.2.3/ -name config.guess | xargs -r chmod u+w; find /home/gly/op-dev-master/build_dir/target-x86_64_musl/qBittorrent-release-4.2.3/ -name config.guess | xargs -r -n1 cp --remove-destination /home/gly/op-dev-master/scripts/config.guess; find /home/gly/op-dev-master/build_dir/target-x86_64_musl/qBittorrent-release-4.2.3/ -name config.sub | xargs -r chmod u+w; find /home/gly/op-dev-master/build_dir/target-x86_64_musl/qBittorrent-release-4.2.3/ -name config.sub | xargs -r -n1 cp --remove-destination /home/gly/op-dev-master/scripts/config.sub; AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/gly/op-dev-master/build_dir/target-x86_64_musl/qBittorrent-release-4.2.3=qBittorrent-release-4.2.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/gly/op-dev-master/build_dir/target-x86_64_musl/qBittorrent-release-4.2.3=qBittorrent-release-4.2.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/gly/op-dev-master/build_dir/target-x86_64_musl/qBittorrent-release-4.2.3=qBittorrent-release-4.2.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/home/gly/op-dev-master/staging_dir/target-x86_64_musl/usr/include -I/home/gly/op-dev-master/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/gly/op-dev-master/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/gly/op-dev-master/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include " LDFLAGS="-L/home/gly/op-dev-master/staging_dir/target-x86_64_musl/usr/lib -L/home/gly/op-dev-master/staging_dir/target-x86_64_musl/lib -L/home/gly/op-dev-master/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/gly/op-dev-master/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro "  QT_QMAKE="/home/gly/op-dev-master/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/bin" PKG_CONFIG_PATH="/home/gly/op-dev-master/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/pkgconfig"  ./configure --target=x86_64-openwrt-linux --host=x86_64-openwrt-linux --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls  --disable-gui --enable-stacktrace=no --with-boost="/home/gly/op-dev-master/staging_dir/target-x86_64_musl/usr" ; fi; )
configure: WARNING: unrecognized options: --disable-nls
configure: loading site script /home/gly/op-dev-master/include/site/x86_64
checking for x86_64-openwrt-linux-gcc... x86_64-openwrt-linux-musl-gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... yes
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether x86_64-openwrt-linux-musl-gcc accepts -g... yes
checking for x86_64-openwrt-linux-musl-gcc option to accept ISO C89... none needed
checking whether x86_64-openwrt-linux-musl-gcc understands -c and -o together... yes
checking whether we are using the GNU C++ compiler... yes
checking whether x86_64-openwrt-linux-musl-g++ accepts -g... yes
checking for a sed that does not truncate output... /home/gly/op-dev-master/staging_dir/host/bin/sed
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-openwrt-linux-gnu
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for x86_64-openwrt-linux-strip... x86_64-openwrt-linux-musl-strip
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports the include directive... yes (GNU style)
checking whether make supports nested variables... yes
checking dependency style of x86_64-openwrt-linux-musl-gcc... none
checking dependency style of x86_64-openwrt-linux-musl-g++... none
checking whether OS is FreeBSD... no
checking whether OS is macOS... no
checking pkg-config is at least version 0.23... yes
[... cut ...]
package/Makefile:111: recipe for target 'package/lean/qBittorrent/compile' failed
make[2]: *** [package/lean/qBittorrent/compile] Error 2
make[2]: Leaving directory '/home/gly/op-dev-master'
package/Makefile:107: recipe for target '/home/gly/op-dev-master/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/gly/op-dev-master/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/gly/op-dev-master'
/home/gly/op-dev-master/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

whalechoi, 2020-07-05:
rblibtorrent 需要升级到1.1.14以上版本支持使用新版boost构建

Lienol, 2020-07-06:
测试没问题了

wjlin2k, 2020-07-06:
可以了，感谢

closed 2020-07-06 by wjlin2k