Lienol/openwrt #352: MT76闭源驱动加入

lexin8, 2020-07-06:
https://github.com/coolsnowwolf/lede/issues/5002#issue-651245686

LEDE今天加入了闭源驱动，开源驱动太难用了，半残废，请求作者也加入吧。

hyird, 2020-07-11:
你先自己先拉过来试试

z826540272, 2020-07-23:
哪里难用  
哪里残废，说清楚

bossoftheworld, 2020-07-25:
> 哪里难用
> 哪里残废，说清楚

确实很难用啊 无线速度比闭源差太多，稳定性也不如

Lienol, 2020-08-02:
我没设备测试啊，你先拉过来试试咯，可以就发个PR

z826540272, 2020-08-14:
> 我没设备测试啊，你先拉过来试试咯，可以就发个PR

闭源驱动bug很多，更改配置会导致没WIFI等问题

AmadeusGhost, 2020-08-15:
> 闭源驱动bug很多，更改配置会导致没WIFI等问题

u1s1，那并不是闭源驱动的 bug，只是因为不是用原生 luci 支持的关系，无线接口会掉罢了

kk3375, 2020-08-18:
> > 我没设备测试啊，你先拉过来试试咯，可以就发个PR
> 
> 闭源驱动bug很多，更改配置会导致没WIFI等问题

新3闭源驱动 2.4G 5G都可以扫描选择无线中继，设置好后2.4G，5G随意切换中继都可以正常上网，非常方便，更改配置也不会出现掉网情况，而且网速信号都非常强，这个闭源驱动中继已经算是优化的非常好了。
链接：https://www.right.com.cn/forum/thread-4041406-1-1.html
[image]

WROIATE, 2020-09-28:
请问现在闭源驱动需要如何选择？

chengyao195, 2020-10-03:
闭源驱动垃圾，为什么这么多人推？

hyird, 2020-10-10:
> 闭源驱动垃圾，为什么这么多人推？
mtk luci的确辣鸡，不过无线本身好一点吧

closed 2021-07-05 by lexin8