Lienol/openwrt #357: dev-19.07实时信息无任何显示

wjlin2k, 2020-07-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：状态>实时信息，负载、流量、连接3项均无任何输出内容

### (2) 路由器型号和固件版本

A：x86-64，固件版本 OpenWrt 19.07-SNAPSHOT / LuCI 17.01 Lienol (git-ba036b3)，内核版本 4.14.187

### (3) 详细日志

A：
[image]

Lienol, 2020-07-10:
[image]
[image]
[image]

wjlin2k, 2020-07-10:
奇怪了，上午怎么试都不行，重启也没用，后来放着不管，下午再看，竟然好了，这问题关了吧，谢谢了。

closed 2020-07-10 by wjlin2k