Lienol/openwrt #363: dev-master minidlna 编译问题

trombonist852, 2020-07-15:
大神，今天早晨同步了一下dev-master的源码然后编译minidlna的时候报错说缺少依赖。
Package minidlna is missing dependencies for the following libraries:
libvorbis.so.0
make[3]: *** [Makefile:88: /home/tr0mb0n1st/op-dev-master/bin/packages/x86_64/packages/minidlna_2019-12-09-0763719f-7_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/tr0mb0n1st/op-dev-master/feeds/packages/multimedia/minidlna'
time: package/feeds/packages/minidlna/compile#0.46#0.22#0.65
make[2]: *** [package/Makefile:113: package/feeds/packages/minidlna/compile] Error 2
make[2]: Leaving directory '/home/tr0mb0n1st/op-dev-master'
make[1]: *** [package/Makefile:107: /home/tr0mb0n1st/op-dev-master/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/tr0mb0n1st/op-dev-master'
make: *** [/home/tr0mb0n1st/op-dev-master/include/toplevel.mk:235：world] 错误 2

Lienol, 2020-07-16:
删掉.config bin tmp
make clean重来

closed 2020-07-16 by Lienol

trombonist852, 2020-07-16:
您说的这些我都试过了，还是显示缺少这个依赖