Lienol/openwrt #375: dev-19.07源码编译arm架构固件，编译过程出现bug。

lxhao61, 2020-07-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
dev-19.07源码编译arm架构固件，编译中出现错误提示，但最后可以编译出固件。此情况反复编译一直存在。

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
### (1) 具体问题
A：dev-19.07源码编译arm架构固件，编译中出现错误提示，但最后可以编译出固件。此情况反复编译一直存在，而采用dev-master源码，无一点问题。
### (2) 路由器型号和固件版本

A：型号：斐讯K3；固件版本：dev-19.07源码编译任何固件版本都一样情况。

### (3) 详细日志
A： 
make[3] -C target/linux install
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
/workdir/openwrt/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 1
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..HEAD
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..a3507aad2086f8674e6f022f5d8f648e6bc727dc
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..HEAD
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..a3507aad2086f8674e6f022f5d8f648e6bc727dc
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..HEAD
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..a3507aad2086f8674e6f022f5d8f648e6bc727dc
 make[1] world
 make[2] target/compile
B:
 make[3] -C target/linux install
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
/workdir/openwrt/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 1
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..HEAD
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..a3507aad2086f8674e6f022f5d8f648e6bc727dc
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..HEAD
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..a3507aad2086f8674e6f022f5d8f648e6bc727dc
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..HEAD
fatal: Invalid revision range ee53a240ac902dc83209008a2671e7fdcf55957a..a3507aad2086f8674e6f022f5d8f648e6bc727dc
make[1]: Entering directory '/workdir/openwrt'
make[2]: Entering directory '/workdir/openwrt'
C:
make[4]: *** [install] Error 2
Makefile:31: recipe for target 'install' failed
make[4]: Leaving directory '/workdir/openwrt/target/linux/bcm53xx'
make[3]: *** [install] Error 2
Makefile:13: recipe for target 'install' failed
make[3]: Leaving directory '/workdir/openwrt/target/linux'
time: target/linux/install#52.04#8.94#71.43
make[2]: *** [target/linux/install] Error 2
target/Makefile:23: recipe for target 'target/linux/install' failed
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [/workdir/openwrt/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.target_install] Error 2
target/Makefile:19: recipe for target '/workdir/openwrt/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.target_install' failed
make[1]: Leaving directory '/workdir/openwrt'
make: *** [world] Error 2
/workdir/openwrt/include/toplevel.mk:225: recipe for target 'world' failed
##[error]Process completed with exit code 2.
虽然可以最后编译出固件，但还是希望大神修复这个bug。谢谢！

Lienol, 2020-08-12:
官源问题，编译BCM53XX直接全部一起编译了

lxhao61, 2020-08-15:
编译BCM53XX直接全部一起编译了，这个是的。但我测试过官方编译，不会报错。

Lienol, 2020-08-17:
solve.

closed 2020-08-17 by Lienol

lxhao61, 2020-08-18:
嗯，是的。谢谢大神的解决！