Lienol/openwrt #376: 星际宝盒cm520-79f网口不能用

CMCCAdmin, 2020-07-27:
dev-master分支,编译出来是5.4.52内核，WAN,LAN口无法使用，luci>网络>接口只有一个lan接口关联了无线，同时uci>网络界面下没有交换机选项。

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.52 (linux@linux-Virtual-Machine) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r13951-2b9f426d8d)) #0 SMP Sat Jul 25 14:07:22 2020
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: MobiPromo CM520-79F
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 130560
[    0.000000]   Normal zone: 1152 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 130560 pages, LIFO batch:31
[    0.000000] percpu: Embedded 15 pages/cpu s30028 r8192 d23220 u61440
[    0.000000] pcpu-alloc: s30028 r8192 d23220 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129408
[    0.000000] Kernel command line: ubi.mtd=rootfs root=mtd:ubi_rootfs rootfstype=squashfs rootwait ubi.mtd=rootfs root=mtd:ubi_rootfs rootfstype=squashfs rootwait ubi.block=0,1 root=/dev/ubiblock0_1
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 508288K/522240K available (5451K kernel code, 181K rwdata, 1444K rodata, 1024K init, 230K bss, 13952K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x2f0/0x498 with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 48.00MHz (virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xb11fd3bfb, max_idle_ns: 440795203732 ns
[    0.000007] sched_clock: 56 bits at 48MHz, resolution 20ns, wraps every 4398046511096ns
[    0.000021] Switching to timer-based delay loop, resolution 20ns
[    0.000223] Calibrating delay loop (skipped), value calculated using timer frequency.. 96.00 BogoMIPS (lpj=480000)
[    0.000242] pid_max: default: 32768 minimum: 301
[    0.000438] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.000458] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.001460] CPU: Testing write buffer coherency: ok
[    0.002445] Setting up static identity map for 0x80300000 - 0x80300060
[    0.002578] rcu: Hierarchical SRCU implementation.
[    0.003004] smp: Bringing up secondary CPUs ...
[    0.006040] smp: Brought up 1 node, 4 CPUs
[    0.006061] SMP: Total of 4 processors activated (384.00 BogoMIPS).
[    0.006069] CPU: All CPU(s) started in SVC mode.
[    0.010607] VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
[    0.010777] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.010803] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.011079] pinctrl core: initialized pinctrl subsystem
[    0.012121] NET: Registered protocol family 16
[    0.012519] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.013715] cpuidle: using governor ladder
[    0.013760] cpuidle: using governor menu
[    0.034422] cryptd: max_cpu_qlen set to 1000
[    0.037812] usbcore: registered new interface driver usbfs
[    0.037887] usbcore: registered new interface driver hub
[    0.037975] usbcore: registered new device driver usb
[    0.038025] pps_core: LinuxPPS API ver. 1 registered
[    0.038034] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    0.038061] PTP clock support registered
[    0.039702] clocksource: Switched to clocksource arch_sys_counter
[    0.040583] thermal_sys: Registered thermal governor 'step_wise'
[    0.040880] NET: Registered protocol family 2
[    0.041485] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 6144 bytes, linear)
[    0.041526] TCP established hash table entries: 4096 (order: 2, 16384 bytes, linear)
[    0.041587] TCP bind hash table entries: 4096 (order: 3, 32768 bytes, linear)
[    0.041677] TCP: Hash tables configured (established 4096 bind 4096)
[    0.041794] UDP hash table entries: 256 (order: 1, 8192 bytes, linear)
[    0.041832] UDP-Lite hash table entries: 256 (order: 1, 8192 bytes, linear)
[    0.042062] NET: Registered protocol family 1
[    0.042102] PCI: CLS 0 bytes, default 64
[    0.044084] workingset: timestamp_bits=14 max_order=17 bucket_order=3
[    0.049609] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[... 287 lines cut ...]
[   32.430584] ath10k_ahb a800000.wifi: unsupported HTC service id: 1536
[   32.431472] ath10k_ahb a800000.wifi: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[   32.436048] ath10k_ahb a800000.wifi: msdu-desc: 2500  skid: 32
[   32.484191] ath10k_ahb a800000.wifi: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[   32.485166] ath10k_ahb a800000.wifi: wmi print 'free: 53204 iram: 13416 sram: 35752'
[   32.818306] ath10k_ahb a800000.wifi: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[   32.831098] br-lan: port 2(wlan1) entered blocking state
[   32.831164] br-lan: port 2(wlan1) entered disabled state
[   32.836002] device wlan1 entered promiscuous mode
[   33.006183] ath10k_ahb a800000.wifi: NOTE:  Firmware DBGLOG output disabled in debug_mask: 0x10000000
[   34.347763] ath10k_ahb a000000.wifi: unsupported HTC service id: 1536
[   34.348220] ath10k_ahb a000000.wifi: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[   34.353308] ath10k_ahb a000000.wifi: msdu-desc: 2500  skid: 32
[   34.403504] ath10k_ahb a000000.wifi: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[   34.403618] ath10k_ahb a000000.wifi: wmi print 'free: 53204 iram: 13416 sram: 35752'
[   34.731021] ath10k_ahb a000000.wifi: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[   34.749862] br-lan: port 3(wlan0) entered blocking state
[   34.749928] br-lan: port 3(wlan0) entered disabled state
[   34.754906] device wlan0 entered promiscuous mode
[   34.995702] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   34.996368] br-lan: port 3(wlan0) entered blocking state
[   35.001351] br-lan: port 3(wlan0) entered forwarding state
[   35.218299] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   35.218924] br-lan: port 2(wlan1) entered blocking state
[   35.223879] br-lan: port 2(wlan1) entered forwarding state

CMCCAdmin, 2020-07-27:
又编译了一遍dev-19.07 这个版本里 luci界面里有了 网口 和交换机选项，但lan口接电脑无法dhcp获取ip地址，wifi正常

maskedeken, 2020-08-01:
在dev-19.07这个分支中，需要对target/linux/ipq40xx/base-files/etc/board.d/02_network做如下修改：

canbox,cm520-79f)
                ucidef_add_switch "switch0" \
                        "0u@eth0" "3:lan:2" "4:lan:1"
                ucidef_set_interface_wan "eth1"
                ;;

closed 2020-08-02 by CMCCAdmin