Lienol/openwrt #381: 你好，能不能增加Openwrt链路聚合proto-bonding这个软件包

LsnmxNB, 2020-07-31:
openwrt-x86-64-generic-squashfs-combined-efi-passwallOpenWRT-2020-07-30.img.gz
OpenWrt SNAPSHOT / LuCI 17.01 Lienol (git-97bc582)
都无法通过opkg update然后安装 proto-bonding 报错啊，大佬请增加一下链路聚合proto-bonding这个软件包

Installing proto-bonding (2020-03-30-1) to root...
Downloading http://downloads.openwrt.org/snapshots/packages/x86_64/packages/proto-bonding_2020-03-30-1_all.ipk
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for proto-bonding:
 * 	kernel (= 5.4.52-1-e7a98746d9b02f285f9fc09fa8df0110)
 * opkg_install_cmd: Cannot install package proto-bonding.

Installing luci-proto-bonding (git-20.126.23183-33df1c1) to root...
Downloading http://downloads.openwrt.org/snapshots/packages/x86_64/luci/luci-proto-bonding_git-20.126.23183-33df1c1_all.ipk
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-proto-bonding:
 * 	kernel (= 5.4.52-1-e7a98746d9b02f285f9fc09fa8df0110)
 * opkg_install_cmd: Cannot install package luci-proto-bonding.

Lienol, 2020-07-31:
自己编译，我不负责编译

closed 2020-07-31 by Lienol

LsnmxNB, 2020-07-31:
@Lienol 好吧，https://t.me/passwallOpenWRT233 发布固件的说来这反馈问题

本本, [31.07.20 19:43]
我只搬运Lienol的actions

本本, [31.07.20 19:43]
你应该pr他的源码和actions
本本, [31.07.20 19:43]
[回复 楚留香 国]
我只搬运Lienol的actions

本本, [31.07.20 19:43]
你应该pr他的源码和actions

本本, [31.07.20 19:45]
合并了，以后发布的就有了