Lienol/openwrt #385: dev-master编译lm-sensors出现错误，望大神帮忙看看，感激不尽！

trombonist852, 2020-08-04:
编译x86固件，luci用的是9.07，因为用17的话有些插件显示会有问题，19显示正常。会跳出如下错误 望大神指点！我试过make clean，make dirclean ，然后删除tmp，config之类的，也重新拉过代码重新编译，还是会跳这个错误，搜过网上的，都没解决....
make[3]: Entering directory '/home/wangjun/op-dev-master/feeds/packages/utils/lm-sensors'
rm -f /home/wangjun/op-dev-master/build_dir/target-x86_64_musl/lm-sensors-3-6-0/.built
touch /home/wangjun/op-dev-master/build_dir/target-x86_64_musl/lm-sensors-3-6-0/.built_check
make  -C /home/wangjun/op-dev-master/build_dir/target-x86_64_musl/lm-sensors-3-6-0 AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/wangjun/op-dev-master/build_dir/target-x86_64_musl/lm-sensors-3-6-0=lm-sensors-3-6-0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size KERNELVERSION="" LINUX="" CC="x86_64-openwrt-linux-musl-gcc" CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/wangjun/op-dev-master/build_dir/target-x86_64_musl/lm-sensors-3-6-0=lm-sensors-3-6-0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CPPFLAGS="-I/home/wangjun/op-dev-master/staging_dir/target-x86_64_musl/usr/include -I/home/wangjun/op-dev-master/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/wangjun/op-dev-master/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/wangjun/op-dev-master/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include" STAGING_DIR="/home/wangjun/op-dev-master/staging_dir/target-x86_64_musl" PREFIX="/usr" MACHINE="x86_64" EXLDFLAGS="" user
make[4]: Entering directory '/home/wangjun/op-dev-master/build_dir/target-x86_64_musl/lm-sensors-3-6-0'
x86_64-openwrt-linux-musl-gcc  -o prog/sensors/sensors prog/sensors/main.ro prog/sensors/chips.ro -liconv -Llib -lsensors
/home/wangjun/op-dev-master/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib/gcc/x86_64-openwrt-linux-musl/8.4.0/../../../../x86_64-openwrt-linux-musl/bin/ld: cannot find -liconv
collect2: error: ld returned 1 exit status
make[4]: *** [prog/sensors/Module.mk:43: prog/sensors/sensors] Error 1
make[4]: Leaving directory '/home/wangjun/op-dev-master/build_dir/target-x86_64_musl/lm-sensors-3-6-0'
make[3]: *** [Makefile:129: /home/wangjun/op-dev-master/build_dir/target-x86_64_musl/lm-sensors-3-6-0/.built] Error 2
make[3]: Leaving directory '/home/wangjun/op-dev-master/feeds/packages/utils/lm-sensors'
time: package/feeds/packages/lm-sensors/compile#0.11#0.01#0.11
make[2]: *** [package/Makefile:113: package/feeds/packages/lm-sensors/compile] Error 2
make[2]: Leaving directory '/home/wangjun/op-dev-master'
make[1]: *** [package/Makefile:107: /home/wangjun/op-dev-master/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/wangjun/op-dev-master'
make: *** [/home/wangjun/op-dev-master/include/toplevel.mk:235：world] 错误 2

xufuren, 2020-08-04:
估计是download阶段有依赖没下载完整，反复download， 一定要看到没有报错，没有404，没有连接失败为止，如果一直404，就到别的站下载来，复制到dl中。

trombonist852, 2020-08-04:
> 估计是download阶段有依赖没下载完整，反复download， 一定要看到没有报错，没有404，没有连接失败为止，如果一直404，就到别的站下载来，复制到dl中。

试过了，没用...

Lienol, 2020-08-12:
x86测试没问题
make clean 试试

closed 2020-08-19 by Lienol