Lienol/openwrt #387: 编译出错误

gabriellliu, 2020-08-07:
### 路由器型号和固件版本

A：k2t

### 详细日志

A：
mkdir -p /home/gabe/op19/staging_dir/hostpkg/stamp
touch /home/gabe/op19/build_dir/hostpkg/Python-3.7.8/.built
touch /home/gabe/op19/staging_dir/hostpkg/stamp/.python3_installed
make[3]: Leaving directory '/home/gabe/op19/feeds/packages/lang/python/python3'
time: package/feeds/packages/python3/host-compile#191.16#33.19#169.26
make[2]: Leaving directory '/home/gabe/op19'
package/Makefile:107: recipe for target '/home/gabe/op19/staging_dir/target-mips_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/gabe/op19/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/gabe/op19'
Build failed - please re-run with -j1 to see the real error message
/home/gabe/op19/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 1

hyird, 2020-08-08:
Build failed - please re-run with -j1 to see the real error message

Lienol, 2020-08-12:
Build failed - please re-run with -j1 to see the real error message

closed 2020-08-17 by Lienol