Lienol/openwrt #390: IPSec VPN 2020.5.5 7baf856 之后又无法上网了

ragouhuo, 2020-08-10:
您好！
默认配置，2020.5.5 7baf856 还正常的，之后至今又不行了。。。
https://github.com/Lienol/openwrt/issues/238，本来问题在这里已解决的。

Lienol, 2020-08-12:
19.07分支测试没问题

ragouhuo, 2020-08-18:
奇怪，我也是 dev-19.07，测试过 NETGEAR R7800和 X86-64，都不正常，苹果手机。可能是基于域名访问的APP无法上网（比如浏览器直接访问网站），基于IP访问的可能正常（比如微信）。

Lienol, 2020-08-18:
估计DNS问题

ragouhuo, 2020-08-18:
刚又编译了 57d3e09，make menuconfig 保持默认，只选了 ipsec ，还是如上所述vpn上网不正常，当然本地LAN上网正常！

ragouhuo, 2020-08-18:
刚又编译了 57d3e09，make menuconfig 保持默认，只选了 ipsec ，还是如上所述vpn上网不正常，当然本地LAN上网正常！x86_64

Michaelhuang123, 2020-08-28:
> 奇怪，我也是 dev-19.07，测试过 NETGEAR R7800和 X86-64，都不正常，苹果手机。可能是基于域名访问的APP无法上网（比如浏览器直接访问网站），基于IP访问的可能正常（比如微信）。

老哥 我的情况跟你一样 能连上软路由后台 但上不去外网 能上微信 你解决了吗

ragouhuo, 2020-08-31:
> > 奇怪，我也是 dev-19.07，测试过 NETGEAR R7800和 X86-64，都不正常，苹果手机。可能是基于域名访问的APP无法上网（比如浏览器直接访问网站），基于IP访问的可能正常（比如微信）。
> 
> 老哥 我的情况跟你一样 能连上软路由后台 但上不去外网 能上微信 你解决了吗

嗯，微信能上。还没解决

Lienol, 2020-11-04:
还是无法复现呢

eearphon, 2021-06-13:
> 还是无法复现呢

昨天试了 1799848，又正常了，老大辛苦了

closed 2021-07-18 by Lienol