Lienol/openwrt #391: 请教一下编译树莓派64位和R2s固件cpu架构为问号的问题

trombonist852, 2020-08-11:
如题请教一下各位大神，树莓派32位就可以正常显示，64位就不行，r2s也是一样，auto_core.diff的补丁也打过了，menuconfig里面autocore也选择了，可就是不显示，请大神们帮忙分析一下是什么原因？是不是还要做一些地方的改动？万分感激！！

wushenghn, 2020-08-11:
今天编译的已经正常显示了

trombonist852, 2020-08-11:
> 我去试试，是dev-master么？还是dev19.07？

Lienol, 2020-08-12:
参考
https://github.com/coolsnowwolf/luci/commit/3310fa416c2db57a9269b7568d9bb2777df7b136

trombonist852, 2020-08-12:
> 参考
> [coolsnowwolf/luci@3310fa4](https://github.com/coolsnowwolf/luci/commit/3310fa416c2db57a9269b7568d9bb2777df7b136)

大神，我用的是您的dev-master分支，因为想试试新的内核，然后feeds.conf.default里面我把src-git luci https://github.com/Lienol/openwrt-luci.git;dev-17.01，这个改成了dev-19.07分支，因为如果用您默认的17.01的话运行aria2没有web-ui和ariang的按钮，menuconfig里我是选中了的，还有一些插件我是编了的，但在路由器界面不显示，19.07就没问题。然后我按照您说的进到modules/luci-mod-admin-full/luasrc/view/admin_status/index.htm里面去看，发现feeds/luci/modules/luci-mod-admin-full/下面只有一个makefile文件，并没有其他文件，我看了一下把feeds.conf.default里面的luci分支从dev-19.07换成dev-18.06就有了。。。您说的让我参考的这个方法是不是只适用于luci的18.06这个分支？我也执行过./scripts/feeds update -a && ./scripts/feeds install -a 这个命令了，请大神赐教啊....

Lienol, 2020-08-12:
只适合17.01和18.06，建议使用。19.07已经改成js，不能使用此方法。
更换feeds后要先
./scripts/feeds clean
再
./scripts/feeds update -a
./scripts/feeds install -a

trombonist852, 2020-08-12:
> 只适合17.01和18.06，建议使用。19.07已经改成js，不能使用此方法。
> 更换feeds后要先
> ./scripts/feeds clean
> 再
> ./scripts/feeds update -a
> ./scripts/feeds install -a

好的大神！谢谢您了！对，每次切换我执行过clean，多谢大神提醒哈！！那18.06如何能集成像19.07里面的软件包的插件？19.07这个超级实用....但是18.06里面就没有....
[image]

trombonist852, 2020-08-12:
> 只适合17.01和18.06，建议使用。19.07已经改成js，不能使用此方法。
> 更换feeds后要先
> ./scripts/feeds clean
> 再
> ./scripts/feeds update -a
> ./scripts/feeds install -a

还有，大神，我刚才git pull了一下dev-master，然后提示 Makefile 'package/system/procd/Makefile' has a dependency on 'libselinux', which does not exist这个错误

Lienol, 2020-08-13:
master分支好正常，跟着官方同步的。

closed 2020-08-19 by Lienol