Lienol/openwrt #398: 4.19.138 编译失败

miaoxinwei, 2020-08-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：2020-08-20T09:27:12.9157884Z /home/runner/work/openwrt_build/openwrt_build/openwrt/build_dir/target-x86_64_musl/linux-x86_64/sched-cake-2019-03-12-057c7388/sch_cake.c:615:6: error: implicit declaration of function 'tc_skb_protocol'; did you mean 'skb_protocol'? [-Werror=implicit-function-declaration]
2020-08-20T09:27:12.9158342Z   if (tc_skb_protocol(skb) != htons(ETH_P_IP))
2020-08-20T09:27:12.9158651Z       ^~~~~~~~~~~~~~~
2020-08-20T09:27:12.9158946Z       skb_protocol

官方解决commit点：https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=c43af7a6cb3303dba044595a6acd8dc8b9d3d046

### (2) 路由器型号和固件版本

A：openwrt

### (3) 详细日志

A：

Lienol, 2020-08-21:
See the latest commit.

miaoxinwei, 2020-08-21:
我验证下