Lienol/openwrt #405: dev-master luci 19.07更新feeds后编译flac提示missing automake-1.16,望大神帮忙看看，感谢

trombonist852, 2020-08-28:
make[4]: Entering directory '/home/wangjun/smp/op-dev-master/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/flac-1.3.3'
 cd . && /bin/bash /home/wangjun/smp/op-dev-master/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/flac-1.3.3/missing automake-1.16 --foreign
/home/wangjun/smp/op-dev-master/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/flac-1.3.3/missing: line 81: automake-1.16: command not found
WARNING: 'automake-1.16' is missing on your system.
         You should only need it if you modified 'Makefile.am' or
         'configure.ac' or m4 files included by 'configure.ac'.
         The 'automake' program is part of the GNU Automake package:
         <https://www.gnu.org/software/automake>
         It also requires GNU Autoconf, GNU m4 and Perl in order to run:
         <https://www.gnu.org/software/autoconf>
         <https://www.gnu.org/software/m4/>
         <https://www.perl.org/>
Makefile:430: recipe for target 'Makefile.in' failed
make[4]: *** [Makefile.in] Error 1
make[4]: Leaving directory '/home/wangjun/smp/op-dev-master/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/flac-1.3.3'
Makefile:79: recipe for target '/home/wangjun/smp/op-dev-master/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/flac-1.3.3/.built' failed
make[3]: *** [/home/wangjun/smp/op-dev-master/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/flac-1.3.3/.built] Error 2
make[3]: Leaving directory '/home/wangjun/smp/op-dev-master/feeds/packages/libs/flac'
time: package/feeds/packages/flac/compile#0.11#0.00#0.11
package/Makefile:111: recipe for target 'package/feeds/packages/flac/compile' failed
make[2]: *** [package/feeds/packages/flac/compile] Error 2
make[2]: Leaving directory '/home/wangjun/smp/op-dev-master'
package/Makefile:107: recipe for target '/home/wangjun/smp/op-dev-master/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile' failed
make[1]: *** [/home/wangjun/smp/op-dev-master/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/wangjun/smp/op-dev-master'
/home/wangjun/smp/op-dev-master/include/toplevel.mk:233: recipe for target 'world' failed
make: *** [world] Error 2

Lienol, 2020-08-28:
make clean
./scripts/feeds clean
./scripts/feeds update -a
./scripts/feeds install -a

closed 2020-08-28 by Lienol

trombonist852, 2020-08-28:
> make clean
> ./scripts/feeds clean
> ./scripts/feeds update -a
> ./scripts/feeds install -a

谢谢大神，这个问题解决了 ，但是刚才在编译的时候碰到passwall、openclash、serverchan这三个软件依赖缺失的错误，这个通过更新feeds能解决不？？还有现在编译好固件之后无法更改设置，比如更改ip地址，或者换主题，都是点一下保存就跳掉，然后右上角就永远会有几项更改未保存的提示
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-openclash:
 * 	curl
 * opkg_install_cmd: Cannot install package luci-app-openclash.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-serverchan:
 * 	curl
 * opkg_install_cmd: Cannot install package luci-app-serverchan.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-passwall:
 * 	curl
 * opkg_install_cmd: Cannot install package luci-app-passwall.
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/wangjun/smp/op-dev-master'
package/Makefile:108: recipe for target '/home/wangjun/smp/op-dev-master/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_install' failed
make[1]: *** [/home/wangjun/smp/op-dev-master/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/wangjun/smp/op-dev-master'
/home/wangjun/smp/op-dev-master/include/toplevel.mk:233: recipe for target 'world' failed