Lienol/openwrt #406: dev-master luci-19.07更新feeds后提示依赖缺失

trombonist852, 2020-08-29:
提示缺少libcurl和curl，然后编译到最后openclash会提示失败，望大神看看，谢谢啦，我已经试过./scripts/feeds clean && ./scripts/feeds update -a && ./scripts/feeds install -a 这三个命令了
WARNING: Makefile 'package/feeds/packages/apache/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/telephony/asterisk-16.x/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/boinc/Makefile' has a dependency on 'curl', which does not exist
WARNING: Makefile 'package/feeds/packages/clamav/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/clamav/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/collectd/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/collectd/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/collectd/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/collectd/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/collectd/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/collectd/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/czmq/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/ddns-scripts/Makefile' has a dependency on 'curl', which does not exist
WARNING: Makefile 'package/feeds/packages/ddns-scripts/Makefile' has a dependency on 'curl', which does not exist
WARNING: Makefile 'package/feeds/packages/ddns-scripts/Makefile' has a dependency on 'curl', which does not exist
WARNING: Makefile 'package/feeds/packages/ddns-scripts/Makefile' has a dependency on 'curl', which does not exist
WARNING: Makefile 'package/feeds/packages/ddns-scripts/Makefile' has a dependency on 'curl', which does not exist
WARNING: Makefile 'package/feeds/packages/ddns-scripts/Makefile' has a dependency on 'curl', which does not exist
WARNING: Makefile 'package/feeds/packages/domoticz/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/elektra/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/esniper/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/forked-daapd/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/freeradius3/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/telephony/freeswitch/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/gammu/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/git/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/gnupg/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/hcxtools/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/hs20/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/https-dns-proxy/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/icecast/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/idevicerestore/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/ipfs-http-client/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/ipfs-http-client/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/telephony/kamailio-5.x/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/telephony/kamailio-5.x/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/telephony/kamailio-5.x/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/telephony/kamailio-5.x/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/leech/Makefile' has a dependency on 'curl', which does not exist
WARNING: Makefile 'package/feeds/packages/libmariadb/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/libnpupnp/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/libulfius/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/libulfius/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/linknx/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/luarocks/Makefile' has a dependency on 'curl', which does not exist
WARNING: Makefile 'package/custom/OpenClash-master/luci-app-openclash/Makefile' has a dependency on 'curl', which does not exist
WARNING: Makefile 'package/feeds/diy1/luci-app-passwall/Makefile' has a dependency on 'curl', which does not exist
WARNING: Makefile 'package/custom/luci-app-serverchan-master/Makefile' has a dependency on 'curl', which does not exist
WARNING: Makefile 'package/feeds/packages/measurement-kit/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/measurement-kit/Makefile' has a build dependency on 'curl', which does not exist
WARNING: Makefile 'package/feeds/packages/mocp/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/mpd/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/mpd/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/mpd/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/netifyd/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/netopeer2/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/noddos/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/ooniprobe/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/openwisp-config/Makefile' has a dependency on 'curl', which does not exist
WARNING: Makefile 'package/feeds/packages/openwisp-config/Makefile' has a dependency on 'curl', which does not exist
WARNING: Makefile 'package/feeds/packages/openwisp-config/Makefile' has a dependency on 'curl', which does not exist
WARNING: Makefile 'package/feeds/packages/openwisp-config/Makefile' has a dependency on 'curl', which does not exist
WARNING: Makefile 'package/feeds/packages/pdns/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/pdns/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/perl-www-curl/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/php7/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/php7-pecl-http/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/python-curl/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/rtorrent/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/rtorrent/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/strongswan/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/syslog-ng/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/transmission/Makefile' has a dependency on 'libcurl', which does not exist
[... cut ...]
WARNING: Makefile 'package/feeds/packages/xmlrpc-c/Makefile' has a dependency on 'libcurl', which does not exist
WARNING: Makefile 'package/feeds/packages/ykclient/Makefile' has a dependency on 'curl', which does not exist
WARNING: Makefile 'package/feeds/packages/ykclient/Makefile' has a build dependency on 'curl', which does not exist
WARNING: Makefile 'package/feeds/packages/ykpers/Makefile' has a dependency on 'curl', which does not exist
WARNING: Makefile 'package/feeds/packages/yubico-pam/Makefile' has a dependency on 'curl', which does not exist

Lienol, 2020-08-29:
你源码坏了）重拉源码重来

trombonist852, 2020-08-29:
> 你源码坏了）重拉源码重来
>
>
>
>
>
>
> 谢谢大神回复！我搞定啦，刚才搜索了一下curl这个文件，对比了之前的文件目录结构，发现是因为packages/network/utils目录下没有curl这个文件夹，我拖了一份之前的curl文件夹进去，发现不会有依赖缺失的提示了，不瞒您说，我从昨天晚上到现在拉了4次代码，不管怎么拉都有这个提示……现在正在重新编，一会儿给您反馈，还有就是用现在的源码编出来树莓派的固件，更改设置之后按保存键无效，比如说换主题或者换iP地址，改好之后没法保存，右上角永远都有未保存的配置这个提示…

trombonist852, 2020-08-29:
> 你源码坏了）重拉源码重来

重新编译后无法保存设置的问题消失，感谢大神！问题关闭

closed 2020-08-29 by trombonist852