Lienol/openwrt #410: cm520-79f编译出现的错误问题

yaak63, 2020-09-02:
CM520-79F固件编译出现以下错误，请教：
1：Makefile:477: recipe for target '~/op19/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openwrt-ipq40xx-generic-canbox_cm520-79f-initramfs-fit-uImage.itb' failed
make[5]: *** [~/op19/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openwrt-ipq40xx-generic-canbox_cm520-79f-initramfs-fit-uImage.itb] Error 1
2：make[1]: *** [~/op19/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.target_install] Error 2
make[1]: Leaving directory '~/op19'
/~/op19/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2
尤其是第二个错误，出现多次，我没有理解‘World’是在哪里。单独编译插件都没有问题。

Lienol, 2020-09-06:
make clean

yaak63, 2020-09-08:
经过make clean 并且执行
./scripts/feeds clean
./scripts/feeds update -a
./scripts/feeds install -a
以后重新编译，仍然失败，问题和从前一样。
Makefile:477: recipe for target '~/op19/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openwrt-ipq40xx-generic-canbox_cm520-79f-initramfs-fit-uImage.itb' failed
make[5]: *** [~/op19/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/tmp/openwrt-ipq40xx-generic-canbox_cm520-79f-initramfs-fit-uImage.itb] Error 1
............
make[1]: *** [~/op19/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.target_install] Error 2
~/op19/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

feeds.conf.default文件为：
src-git packages https://github.com/Lienol/openwrt-packages.git;dev-19.07
src-git routing https://git.openwrt.org/feed/routing.git;openwrt-19.07
src-git telephony https://git.openwrt.org/feed/telephony.git;openwrt-19.07
src-git luci https://github.com/Lienol/openwrt-luci.git;dev-17.01
src-git diy1 https://github.com/xiaorouji/openwrt-package.git;master
src-git kenzo https://github.com/kenzok8/openwrt-packages
其实，上次编译是直接拉取仓库以后就直接编译的，应该是没有被混淆的。请多指教。

closed 2020-11-01 by Lienol