Lienol/openwrt #412: 今天更新feed后提示依赖缺失，导致编译失败。

git-icu, 2020-09-02:
Updating feed 'diy1' from 'https://github.com/xiaorouji/openwrt-package.git;master' ...
已经是最新的。
Create index file './feeds/diy1.index' 
Updating feed 'freifunk' from 'https://github.com/freifunk/openwrt-packages.git' ...
已经是最新的。
Create index file './feeds/freifunk.index' 
WARNING: Makefile 'package/feeds/packages/apache/Makefile' has a dependency on 'libnghttp2', which does not exist
WARNING: Makefile 'package/feeds/packages/curl/Makefile' has a dependency on 'libnghttp2', which does not exist
WARNING: Makefile 'package/feeds/packages/node/Makefile' has a dependency on 'libnghttp2', which does not exist
WARNING: Makefile 'package/feeds/packages/snort/Makefile' has a dependency on 'libnghttp2', which does not exist
Installing all packages from feed packages.
Installing all packages from feed routing.
Installing all packages from feed telephony.
Installing all packages from feed luci.
Installing all packages from feed diy1.
Installing all packages from feed freifunk.

Lienol, 2020-09-02:
rm -rf bin tmp package/feeds
git pull
./scripts/feeds update -a
./scripts/feeds install -a

git-icu, 2020-09-02:
> rm -rf bin tmp軟件包/提要
> git pull
> ./scripts/feeds更新-a
> ./scripts/feeds install -a

感谢！ 已经编译成功~！可以关闭issue了。