Lienol/openwrt #413: ttytd

trombonist852, 2020-09-03:


lysoul147, 2020-09-04:
同样的问题

git-icu, 2020-09-04:
同样的问题,好像netdate打开也是空白的。

viagram, 2020-09-04:
https不能加载http
`

<div class="cbi-map">
--
  | <h2 name="content">TTYD 终端</h2>
  | <iframe id="terminal" style="width: 100%; min-height: 500px; border: none; border-radius: 3px;"></iframe>
  | </div>
  | <script type="text/javascript">
  | document.getElementById("terminal").src = "http://" + window.location.hostname + ":7681";
  | </script>

`

trombonist852, 2020-09-04:
> https不能加载http
> 
> > 
> 
> ```
>     <h2 name="content">TTYD 终端</h2>
>     <iframe id="terminal" style="width: 100%; min-height: 500px; border: none; border-radius: 3px;"></iframe>
> ```
> 
> <script type="text/javascript"> document.getElementById("terminal").src = "http://" + window.location.hostname + ":7681"; </script>

麻烦问一下哈，这个要在哪里修改？？

viagram, 2020-09-04:
> 麻烦问一下哈，这个要在哪里修改？？

关闭https，用http访问路由器

Lienol, 2020-09-06:
楼上正解

closed 2020-09-06 by Lienol

trombonist852, 2020-09-06:
> 楼上正解

大神，还有个问题，新的源码编出来r8000没有WiFi…之前是有的

Lienol, 2020-09-10:
尝试点击 配置 - 接口选br-lan

trombonist852, 2020-09-13:
> 尝试点击 配置 - 接口选br-lan 然后保存

试过啦大神，没用，我看您同步了源码，我重拉源码重新编试试看