Lienol/openwrt #426: samba挂载和速度问题

dache123, 2020-09-17:
硬件是CM520-79F，就是这两天编译的固件，折腾了好久samba挂载ntfs格式的硬盘，就是挂载上去了，电脑没有权限读取，我以为是设置的问题，后来搜索了好久折腾了好久，原来是ntfs挂载的话只能只读挂载，要通过ntfs-3G挂载才可以，固件里默认是都安装了，但是固件默认是通过ntfs挂载，也就导致一直无法用电脑连接samba。后来删除ntfs就好了。
好了之后后续问题又来了

速度，速度问题。

正常通过windows往路由挂载的硬盘里存储文件，速度只有5~6M左右，还是网线连接的。然后我又刷如lede固件，读写速度20M+

这个速度这个是我没选择什么驱动，模块之类的吗？为何差异这么大。

lexin8, 2020-09-18:
硬路由USB挂着硬盘问题很多，驱动、供电、移动硬盘、硬件运行等等都会影响。
[image]

dache123, 2020-09-19:
> 硬路由USB挂着硬盘问题很多，驱动、供电、移动硬盘、硬件运行等等都会影响。
> [image]

你这速度也太刺激了。。。想必是软路由？实不相瞒，我硬盘里复制都没这速度啊。。。哈哈

hyird, 2020-10-10:
ntfs-3g驱动拉垮吧 大雕用的是这个antfs-mount

dache123, 2020-10-25:
是的,换了antfs就好了

closed 2020-11-01 by Lienol