Lienol/openwrt #428: Address in use 日志报错什么问题？

lexin8, 2020-09-18:
Fri Sep 18 17:30:32 2020 user.warn igmpproxy[7111]: MRT_ADD_MEMBERSHIP failed; Errno(98): Address in use
Fri Sep 18 17:31:04 2020 user.warn igmpproxy[7111]: MRT_ADD_MEMBERSHIP failed; Errno(98): Address in use
Fri Sep 18 17:31:32 2020 user.warn igmpproxy[7111]: MRT_ADD_MEMBERSHIP failed; Errno(98): Address in use
Fri Sep 18 17:31:33 2020 user.warn igmpproxy[7111]: MRT_ADD_MEMBERSHIP failed; Errno(98): Address in use
Fri Sep 18 17:31:37 2020 user.warn igmpproxy[7111]: MRT_ADD_MEMBERSHIP failed; Errno(98): Address in use
Fri Sep 18 17:31:41 2020 user.warn igmpproxy[7111]: MRT_ADD_MEMBERSHIP failed; Errno(98): Address in use
Fri Sep 18 17:31:44 2020 user.warn igmpproxy[7111]: MRT_ADD_MEMBERSHIP failed; Errno(98): Address in use
Fri Sep 18 17:31:52 2020 user.warn igmpproxy[7111]: MRT_ADD_MEMBERSHIP failed; Errno(98): Address in use
Fri Sep 18 17:32:05 2020 user.warn igmpproxy[7111]: MRT_ADD_MEMBERSHIP failed; Errno(98): Address in use
Fri Sep 18 17:32:31 2020 user.warn igmpproxy[7111]: MRT_ADD_MEMBERSHIP failed; Errno(98): Address in use
Fri Sep 18 17:33:02 2020 user.warn igmpproxy[7111]: MRT_ADD_MEMBERSHIP failed; Errno(98): Address in use
Fri Sep 18 17:33:25 2020 user.warn igmpproxy[7111]: MRT_ADD_MEMBERSHIP failed; Errno(98): Address in use
Fri Sep 18 17:33:33 2020 user.warn igmpproxy[7111]: MRT_ADD_MEMBERSHIP failed; Errno(98): Address in use
Fri Sep 18 17:33:43 2020 user.warn igmpproxy[7111]: MRT_ADD_MEMBERSHIP failed; Errno(98): Address in use
Fri Sep 18 17:33:46 2020 user.warn igmpproxy[7111]: MRT_ADD_MEMBERSHIP failed; Errno(98): Address in use
Fri Sep 18 17:34:33 2020 user.warn igmpproxy[7111]: MRT_ADD_MEMBERSHIP failed; Errno(98): Address in use
Fri Sep 18 17:34:50 2020 user.warn igmpproxy[7111]: MRT_ADD_MEMBERSHIP failed; Errno(98): Address in use
Fri Sep 18 17:35:32 2020 user.warn igmpproxy[7111]: MRT_ADD_MEMBERSHIP failed; Errno(98): Address in use

Lienol, 2020-09-20:
端口被占用了吧

lexin8, 2020-09-20:
> 端口被占用了吧
现在每天都会异常自动重启，不知道跟这个有没有关系！端口占用没看出来。
[image]

[image]
[image]

closed 2020-11-01 by Lienol