Lienol/openwrt #429: 使用nginx时会导致界面保存失效

WROIATE, 2020-09-18:
详细如图，保存的reload操作进程会一直无法执行，换回uhttpd后恢复正常
[image]
[image]

WROIATE, 2020-09-18:
现象是出现这个情况后在luci当中点任何保存应用都会失效，会直接显示 保存应用->配置完成，而实际上并没有重启应用，正常应该是 保存应用->/etc/init.d/某文件运行->配置完成 ，不清楚是否是nginx原因，目前切换回uhttpd暂时无问题

Lienol, 2020-09-20:
用的是哪个分支

WROIATE, 2020-09-20:
> 用的是哪个分支

dev-19.07 x86_64
这个问题我在lean的issue里也发现了 [issue](https://github.com/coolsnowwolf/lede/issues/4288#issue-598442869)

WROIATE, 2020-09-29:
由于不知道为什么锁无法释放，我在那个issue的基础上改了一下，看下这个解决方案如何，我在本地测试正常
```
lock "/var/run/luci-reload"

config_load ucitrack

for i in $*; do
        config_foreach apply_config $i $i
        echo $i >> /root/log
done
rm -f "/var/run/luci-reload-status"
pid=`cat /var/run/luci-reload` #获取pid
lock -u "/var/run/luci-reload" && kill -9 $pid #在释放锁后杀掉进程
```

VergilGao, 2020-11-15:
19.07 也遇到了这个问题

VergilGao, 2020-11-15:
> 由于不知道为什么锁无法释放，我在那个issue的基础上改了一下，看下这个解决方案如何，我在本地测试正常
>
> ```
> lock "/var/run/luci-reload"
>
> config_load ucitrack
>
> for i in $*; do
>         config_foreach apply_config $i $i
>         echo $i >> /root/log [...]

这只能算是临时的解决方案吧，重点是找到锁没有释放的原因。

WROIATE, 2020-11-16:
> 这只能算是临时的解决方案吧，重点是找到锁没有释放的原因。

是的...如果我知道是为啥我就PR了

VergilGao, 2020-11-16:
> 是的...如果我知道是为啥我就PR了

>.< 不知道能不能打断点调试什么的，我这单纯的输出命令结果也看不出来什么来，就是 lock -u 在某种情况下没有生效（这条语句之前和之后的语句都生效了），luci调用完之后手动 lock -u 也不行。
我试着用18.06版本的luci界面编译一下看看先。

WROIATE, 2020-11-16:
@VergilGao 应该是和nginx有关的问题，而且不知道为啥lienol的nginx无法使用http2，官方的倒是一切正常

VergilGao, 2020-11-16:
> @VergilGao 应该是和nginx有关的问题，而且不知道为啥lienol的nginx无法使用http2，官方的倒是一切正常

官方的19.07和这个仓库有什么区别？如果我要用官方版本编译的话，和用这个仓库编译有什么不同的地方？因为我实际上用到的软件包很少，核心需求就是 kms openclash ddns smartdns  nginx是刚需，我买这个工控机路由就是为了nginx，再就是zsh boybu 之类的常用的shell工具。

Lienol, 2020-11-17:
更新最新的源码和feeds再试试

VergilGao, 2020-11-17:
> 更新最新的源码和feeds再试试

啊这……多线程的确容易出现死锁，但是因此直接改成单线程也太赞了吧……

WROIATE, 2020-11-18:
@VergilGao 解决了我就关掉issye了

closed 2020-11-18 by WROIATE