Lienol/openwrt #434: 怎样去掉概况网络状态ipv6情况

pangfeng161, 2020-09-26:
编译怎样去掉概况网络状态ipv6情况,用不到ipv6看着影响,leda源码编译没有这界面

closed 2020-10-02 by Lienol

qq1316237299, 2020-10-08:
> 编译怎样去掉概况网络状态ipv6情况,用不到ipv6看着影响,leda源码编译没有这界面

兄弟解决了吗，我也想去除掉

Lienol, 2020-10-09:
https://github.com/Lienol/openwrt-luci/commit/df6464190b82faacf4eb29d62d100235846bf437

qq1316237299, 2020-10-09:
> > [Lienol / openwrt-luci @ df64641](https://github.com/Lienol/openwrt-luci/commit/df6464190b82faacf4eb29d62d100235846bf437)
> 
> 谢谢li大

li大这是更新的github上的吗

qq1316237299, 2020-10-11:
[image]
[image]
大佬我没编译ipv6helper 怎么会有ipv6的地址的，还可以使用，只能把这个ipv6解析关了，才好，希望大佬可以从源码把ipv6清除了，谢谢