Lienol/openwrt #435: 5.4.67无法用LEDE Luci吗？交换机界面打不开?

xiaoxifu64, 2020-09-27:
5.4.67无法用LEDE Luci吗？交换机界面打不开
报错如下
/usr/lib/lua/luci/dispatcher.lua:469: Failed to execute cbi dispatcher target for entry '/admin/network/vlan'.
The called action terminated with an exception:
/usr/lib/lua/luci/model/cbi/admin_network/vlan.lua:190: bad argument #2 to 'tonumber' (integer expected, got string)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:469: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>

其他页面正常，有解决办法吗？

Lienol, 2020-09-27:
你确定用我的源码编译的？

xiaoxifu64, 2020-09-27:
确定，我改了下LUCI 
src-git packages https://github.com/Lienol/openwrt-packages.git;dev-master
src-git routing https://git.openwrt.org/feed/routing.git
src-git telephony https://git.openwrt.org/feed/telephony.git
src-git luci https://github.com/Lienol/openwrt-luci.git;dev-17.01
src-git diy1 https://github.com/xiaorouji/openwrt-package.git;master
src-git freifunk https://github.com/freifunk/openwrt-packages.git

xiaoxifu64, 2020-09-27:
因为之前用过你的源码  5.4.61 LUCI 版本是17.01 最新的5.4.67 改成17.01就用不了

closed 2020-10-02 by Lienol