Lienol/openwrt #436: 无法上国内网站，但google可以

rangerwill, 2020-09-27:
1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)

    没有类似的issue

2. 详细叙述

斐讯n1和x64软路由都出现无法上国内网站，google等境外网站能正常访问的情况。
如果手动重启防火墙，则可以正常上网。
但过几小时之后，又不行了。
(1) 具体问题

A：斐讯n1和x64软路由都出现无法上国内网站，google等境外网站能正常访问的情况。
如果手动重启防火墙，则可以正常上网。
但过几小时之后，又不行了。
(2) 路由器型号和固件版本

A：
1.斐讯N1
OpenWrt R20.9.15 / LuCI Master (git-20.223.26773-d18ef13)
2.Intel J3710 + 8G DDR3
OpenWrt R20.9.15 / LuCI Master (git-20.256.12360-1a54222)
3.Intel i5 6400 + 8G DDR4
OpenWrt R20.9.15 / LuCI Master (git-20.256.12360-1a54222)
(3) 详细日志

A：暂时无法提供

lexin8, 2020-09-27:
Node problem

rangerwill, 2020-09-29:
> 
> 
> Node problem

obviously that is not node problem
The problem occured when I was using v2ray node which is constucted by myself
And it also occured when I was using ssr node which is provided by SSR providers.
I assume that it couldn't be a node problem.

rangerwill, 2020-09-30:
这个问题是 最近半个月的新版本出现的 半个月以上的老版本没这个问题

rangerwill, 2020-10-02:
[image]

closed 2020-10-02 by Lienol

rangerwill, 2020-11-02:
service mwan3 stop
service mwan3 disable

done.