Lienol/openwrt #441: unbound-anchor编译错误

Dimitri2020007, 2020-10-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：x86_64

### (3) 详细日志

A：

install -m0755 /home/dimitri/openwrt/build_dir/target-x86_64_musl/unbound-heavy/unbound-1.11.0/ipkg-install/usr/sbin/unbound-anchor /home/dimitri/openwrt/build_dir/target-x86_64_musl/unbound-heavy/unbound-1.11.0/ipkg-x86_64/unbound-anchor/usr/sbin/
find /home/dimitri/openwrt/build_dir/target-x86_64_musl/unbound-heavy/unbound-1.11.0/ipkg-x86_64/unbound-anchor -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
Package unbound-anchor is missing dependencies for the following libraries:
libevent-2.1.so.7
make[3]: *** [Makefile:267: /home/dimitri/openwrt/bin/packages/x86_64/packages/unbound-anchor_1.11.0-2_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/dimitri/openwrt/feeds/packages/net/unbound'
time: package/feeds/packages/unbound/heavy/compile#45.54#7.58#52.59
make[2]: *** [package/Makefile:119: package/feeds/packages/unbound/compile] Error 2
make[2]: Leaving directory '/home/dimitri/openwrt'
make[1]: *** [package/Makefile:107: /home/dimitri/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/dimitri/openwrt'
make: *** [/home/dimitri/openwrt/include/toplevel.mk:227: world] Error 2

Lienol, 2020-10-07:
选上 libevent 编译试试

Dimitri2020007, 2020-10-09:
这个已经解决了，可以关闭问题了。

closed 2020-10-10 by Lienol