Lienol/openwrt #443: wsdd2

kl9800, 2020-10-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
请看日志

### (1) 具体问题

A：编译的时候遇到wsdd2问题，请问我现在应该如何操作让编译可以继续下去？

### (2) 路由器型号和固件版本

A：X86

### (3) 详细日志

A：Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for autosamba:
 * 	**wsdd2**
 * opkg_install_cmd: Cannot install package autosamba.
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/xpp/op19'
package/Makefile:108: recipe for target '/home/xpp/op19/staging_dir/target-x86_64_musl/stamp/.package_install' failed
make[1]: *** [/home/xpp/op19/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/xpp/op19'
/home/xpp/op19/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2

Lienol, 2020-10-07:
rm -rf bin tmp package/feeds
git pull
./scripts/feeds update -a
./scripts/feeds install -a

kl9800, 2020-10-07:
谢谢

closed 2020-10-07 by kl9800