Lienol/openwrt #446: 编译luci-app-filetransfer时遇到错误，望各位大神出个招，感激不尽

trombonist852, 2020-10-09:
最近更新到最新源码后编译luci-app-filetransfer时会跳出
wangjun@Trb:~/openwrt$ make package/custom/luci-app-filetransfer/{clean,compile} V=s
time: target/linux/prereq#0.06#0.01#0.08
make[1]: Entering directory '/home/wangjun/openwrt'
make[2]: Entering directory '/home/wangjun/openwrt/package/custom/luci-app-filetransfer'
/home/wangjun/openwrt/feeds/luci/luci.mk:305: *** Package/luci-i18n-filetransfer-zh-cn is missing the VERSION field. Stop.
make[2]: Leaving directory '/home/wangjun/openwrt/package/custom/luci-app-filetransfer'
time: package/custom/luci-app-filetransfer/clean#0.80#0.02#0.82
package/Makefile:111: recipe for target 'package/custom/luci-app-filetransfer/clean' failed
make[1]: *** [package/custom/luci-app-filetransfer/clean] Error 2
make[1]: Leaving directory '/home/wangjun/openwrt'
/home/wangjun/openwrt/include/toplevel.mk:234: recipe for target 'package/custom/luci-app-filetransfer/clean' failed
make: *** [package/custom/luci-app-filetransfer/clean] Error 2

这个问题去谷歌了一圈也没搜出个所以然，望大神们出个招，这个是啥问题导致的？？感激不尽！

hyird, 2020-10-10:
先编一个固件出来 这个单独ipk编译 一开始就上就是报错 我也没查到解决办法
如果只想单独编译ipk 推荐用SDK

trombonist852, 2020-10-10:
我昨天找到解决办法啦，
https://github.com/openwrt/luci/commit/e6f77d5d72c0e13ed3eb370419abab52c4920a32
手动把这个补丁打到luci.mk和luci.js里面，就好了，你试试

closed 2020-10-11 by trombonist852