Lienol/openwrt #447: openwrt-actions的问题可以修复吗

ldsj007, 2020-10-10:
Check warning on line 1 in .github

@github-actions
github-actions
/ Build x64-1907

.github#L1
The `set-env` command is deprecated and will be disabled soon. Please upgrade to using Environment Files. For more information see: https://github.blog/changelog/2020-10-01-github-actions-deprecating-set-env-and-add-path-commands/
 Check failure on line 1 in .github

@github-actions
github-actions
/ Build x64-1907

.github#L1
Process completed with exit code 2.

45m 57s
1
Run cd openwrt
19
3 thread compile
20
 make[1] world
21
 make[2] tools/compile
22
 make[2] package/cleanup
23
 make[3] -C tools/flock compile
24
 make[3] -C tools/xz compile
25
 make[3] -C tools/sed compile
26
 make[3] -C tools/patch compile
27
 make[3] -C tools/tar compile
28
 make[3] -C tools/pkg-config compile
29
 make[3] -C tools/m4 compile
30
 make[3] -C tools/expat compile
31
 make[3] -C tools/sstrip compile
32
 make[3] -C tools/patch-image compile
33
 make[3] -C tools/scons compile
34
 make[3] -C tools/mtools compile
35
 make[3] -C tools/ucl compile
36
 make[3] -C tools/lzma compile
37
 make[3] -C tools/zip compile
38
 make[3] -C tools/autoconf compile
39
 make[3] -C tools/missing-macros compile
40
 make[3] -C tools/automake compile
41
 make[3] -C tools/libressl compile
42
 make[3] -C tools/libtool compile
43
 make[3] -C tools/dosfstools compile
[... 187 lines cut ...]
make[4]: Leaving directory '/home/runner/work/openwrt-actions/openwrt-actions/openwrt/target/linux/x86'
138
Makefile:13: recipe for target 'compile' failed
139
make[3]: Leaving directory '/home/runner/work/openwrt-actions/openwrt-actions/openwrt/target/linux'
140
time: target/linux/compile#1.43#0.59#2.33
141
target/Makefile:23: recipe for target 'target/linux/compile' failed
142
make[2]: Leaving directory '/home/runner/work/openwrt-actions/openwrt-actions/openwrt'
143
make[2]: *** [target/linux/compile] Error 2
144
make[1]: *** [/home/runner/work/openwrt-actions/openwrt-actions/openwrt/staging_dir/target-x86_64_musl/stamp/.target_compile] Error 2
145
target/Makefile:18: recipe for target '/home/runner/work/openwrt-actions/openwrt-actions/openwrt/staging_dir/target-x86_64_musl/stamp/.target_compile' failed
146
make[1]: Leaving directory '/home/runner/work/openwrt-actions/openwrt-actions/openwrt'
147
/home/runner/work/openwrt-actions/openwrt-actions/openwrt/include/toplevel.mk:225: recipe for target 'world' failed
148
make: *** [world] Error 2
149
Error: Process completed with exit code 2.

Lienol, 2020-10-10:
重新跑一遍试试

Dimitri2020007, 2020-10-10:
这个要改actions脚本，跟固件源码没啥问题。

脚本源适配了官方openwrt和lean openwrt源码，让action适配这个openwrt建议从出错的地方下手。

先从set-env下手

ldsj007, 2020-10-10:
你们看下跟这个有关系吗？
这是报错里面给出的详情链接
https://github.blog/changelog/2020-10-01-github-actions-deprecating-set-env-and-add-path-commands/

ldsj007, 2020-10-10:
> 这个要改actions脚本，跟固件源码没啥问题。
> 
> 脚本源适应了官方的openwrt和lean openwrt二进制文件，让动作适应这个openwrt建议从出错的地方下手。
> 
> 另外你脚本出错的例程是不是在1小时左右（总过程，脚本错误在编译是40分钟左右的位置）
> 
> 如果是这种还真的不好弄
> 
> 我曾用的是这个脚本https://github.com/P3TERX/Actions-OpenWrt.git
> 
> 总是1小时左右就失败了，实体编译没有问题。

我之前Lienol
/
openwrt-actions是可以编译的，就前几天才不行了，
出错时间不确定了，
P3TERX/Actions-OpenWrt.git我还没用

Dimitri2020007, 2020-10-10:
你用的那个脚本源是不是改分支了。
最近好多源码都改分支了
所以编译失败了

ldsj007, 2020-10-10:
你用的那个脚本源是不是改分支了。
-----分支名称我应该已经改了

昨天Lienol更新了脚本，我是今天fork的自动编译脚本，到我那里改了点机型信息就不行，而昨天在Lienol的原始库还有正常运行的记录

Dimitri2020007, 2020-10-10:
应该是后者的问题

hyird, 2020-10-11:
内核config 少个东西而已
nables kernel network stack path for Shortcut Forwarding Engine (SHORTCUT_FE) [N/y] (NEW) aborted
要按下回车才能过。
已经修复了 
重新跑下就行

ldsj007, 2020-10-11:
> 内核config 少个东西而已
> nables kernel network stack path for Shortcut Forwarding Engine (SHORTCUT_FE) [N/y] (NEW) aborted
> 要按下回车才能过。
> 已经修复了
> 重新跑下就行

--------------------------------------------------------------------------------------------------
能得到你们的帮助实在是太棒了，。

黄色警报提示的这个
 `set-env`
能用别的指令代替吗？
没有警报会提升逼格200%  ！

Dimitri2020007, 2020-10-14:
这个脚本已经更新可以去试一下 https://github.com/P3TERX/Actions-OpenWrt.git

ldsj007, 2020-10-14:
好的，谢谢啦。

closed 2020-10-14 by ldsj007