Lienol/openwrt #450: 自有vps服务器，如何实现编译的openwrt固件定期版本更新

channingbox, 2020-10-10:
如题，当客户端检查到vps服务器端有新版本openwrt固件时，自动实现固件更新又不丢失配置文件。

hyird, 2020-10-11:
这属于你自己的问题，发issue过分了吧

channingbox, 2020-10-11:
> 这属于你自己的问题，发issue过分了吧

还好吧，就是想和大家探讨一下解决方案，不懂才问的啊，这里不能发的话真的不知道到哪里去问。

viagram, 2020-10-11:
我目前是这样：
1，vps上每晚检测一次openwrt和package有没有更新，有就github自动编译，然后上传到web下。
2，路由上自写一个检测脚本，每天上早上检测web上有没有新版，有就自动下载更新。
用了快一年了，没什么大问题。

channingbox, 2020-10-11:
> 我目前是这样：
> 1，vps上每晚检测一次openwrt和package有没有更新，有就github自动编译，然后上传到web下。
> 2，路由上自写一个检测脚本，每天上早上检测web上有没有新版，有就自动下载更新。
> 用了快一年了，没什么大问题。

求大神在github开一个project，可以让小白可以依葫芦画瓢的。

qq1316237299, 2020-10-12:
> > 我目前是这样：
> > 1，vps上每晚检测一次openwrt和package有没有更新，有就github自动编译，然后上传到web下。
> > 2，路由上自写一个检测脚本，每天上早上检测web上有没有新版，有就自动下载更新。
> > 用了快一年了，没什么大问题。
> 
> 求大神在github开一个project，可以让小白可以依葫芦画瓢的。不过个人觉得每天检测频率太高了，10天或半个月检测一次并且编译更新就可以。

github又不占用你的资源，手动滑稽

lylus, 2020-10-19:
> 我目前是这样：
> 1，vps上每晚检测一次openwrt和package有没有更新，有就github自动编译，然后上传到web下。
> 2，路由上自写一个检测脚本，每天上早上检测web上有没有新版，有就自动下载更新。
> 用了快一年了，没什么大问题。

我也想这样弄，后来想着万一新编译的有某个严重bug，这样就完蛋了。

channingbox, 2020-10-21:
可能某一个版本存在bug，但不至于出现严重的bug，现在固件已经这么成熟了，即使有小bug，下个版本也会及时修复。万一出现不能启动的bug，也可手动编译恢复，家庭使用能有什么严重的后果。

hyird, 2020-10-21:
> 可能某一个版本存在bug，但不至于出现严重的bug，现在固件已经这么成熟了，即使有小bug，下个版本也会及时修复。万一出现不能启动的bug，也可手动编译恢复，家庭使用能有什么严重的后果。

打扰 媳妇追剧，下个月零花钱没了

closed 2020-11-01 by Lienol