Lienol/openwrt #452: 大佬我没编译ipv6helper 怎么会有ipv6的地址的，还可以使用，只能把这个ipv6解析关了，才好，希望大佬可以从源码把ipv6彻底清除了，谢谢

qq1316237299, 2020-10-11:
[image]
[image]

qq1316237299, 2020-10-11:
[image]
这个就是取消不了

FreeJ233, 2020-10-15:
如果不想要ipv6 在 Global build settings 里面取消ipv6勾选就可以。

qq1316237299, 2020-10-15:
> 如果不想要ipv6在全局构建设置里面取消ipv6重置就可以。
在哪个选项里呢 大佬
[image]

qq1316237299, 2020-10-15:
> 如果不想要ipv6 在 Global build settings 里面取消ipv6勾选就可以。

[image]
大佬 这个lean的源码里好像也是默认勾选的， 我这里试过取消掉会编译失败，没编译失败的话 docker也会用不了。。。

pangfeng161, 2020-10-16:
> > 如果不想要ipv6在全局构建设置里面取消ipv6重置就可以。
> 
> [image]
> 大佬这个lean的原始代码里好像也是替代的，，我这里试过取消取消会编译失败，没编译失败的话docker也会用不了。。。

解决了没有,我这编译成功但不能拨号

Lienol, 2020-11-04:
拨号时可手动关闭ipv6

closed 2020-11-04 by Lienol