Lienol/openwrt #455: 云编译后无法使用Web管理页面

aheig, 2020-10-13:
浏览器访问Web管理地址，返回下面的内容

/usr/lib/lua/luci/model/cbi/passwall/api/v2ray.lua:6: module 'luci.model.ipkg' not found:
	no field package.preload['luci.model.ipkg']
	no file './luci/model/ipkg.lua'
	no file '/usr/share/lua/luci/model/ipkg.lua'
	no file '/usr/share/lua/luci/model/ipkg/init.lua'
	no file '/usr/lib/lua/luci/model/ipkg.lua'
	no file '/usr/lib/lua/luci/model/ipkg/init.lua'
	no file './luci/model/ipkg.so'
	no file '/usr/lib/lua/luci/model/ipkg.so'
	no file '/usr/lib/lua/loadall.so'
	no file './luci.so'
	no file '/usr/lib/lua/luci.so'
	no file '/usr/lib/lua/loadall.so'
stack traceback:
	[C]: in function 'require'
	/usr/lib/lua/luci/model/cbi/passwall/api/v2ray.lua:6: in main chunk
	[C]: in function 'require'
	/usr/lib/lua/luci/controller/passwall.lua:9: in main chunk
	[C]: in function 'require'
	/usr/lib/lua/luci/dispatcher.lua:1105: in function 'createindex'
	/usr/lib/lua/luci/dispatcher.lua:1205: in function 'createtree'
	/usr/lib/lua/luci/dispatcher.lua:695: in function 'menu_json'
	/usr/lib/lua/luci/dispatcher.lua:796: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:479: in function </usr/lib/lua/luci/dispatcher.lua:478>

lexin8, 2020-10-19:
用Lienol源码，不是大神就别去搞官方源码。

aheig, 2020-10-22:
还真没使用其他源码，也许是减了DDNS插件的原因吧

Lienol, 2020-11-01:
选上luci-lib-ipkg编译

closed 2020-11-01 by Lienol