Lienol/openwrt #465: cm520-79f ipq4019应用vlan补丁出错715-net-essedma-disable-default-vlan.patch

CMCCAdmin, 2020-11-02:
Hunk #6 succeeded at 1110 (offset -4 lines).
Hunk #7 succeeded at 1135 (offset -4 lines).
Hunk #8 succeeded at 1259 (offset -2 lines).
Hunk #9 succeeded at 1307 (offset -2 lines).

Applying /home/ubuntu/Desktop/cm520/openwrt/target/linux/ipq40xx/patches-5.4/715-net-essedma-disable-default-vlan.patch using plaintext: 
patching file arch/arm/boot/dts/qcom-ipq4019.dtsi
Hunk #1 succeeded at 648 (offset 17 lines).
Hunk #2 succeeded at 686 (offset 17 lines).
patching file drivers/net/ethernet/qualcomm/essedma/edma.c
Hunk #1 succeeded at 719 (offset -10 lines).
Hunk #2 FAILED at 1412.
1 out of 2 hunks FAILED -- saving rejects to file drivers/net/ethernet/qualcomm/essedma/edma.c.rej
Patch failed!  Please fix /home/ubuntu/Desktop/cm520/openwrt/target/linux/ipq40xx/patches-5.4/715-net-essedma-disable-default-vlan.patch!
make[4]: *** [Makefile:34: /home/ubuntu/Desktop/cm520/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/linux-5.4.72/.prepared_e9d460ceee40f460ec448a94c4a80cb4] Error 1
make[4]: Leaving directory '/home/ubuntu/Desktop/cm520/openwrt/target/linux/ipq40xx'
make[3]: *** [Makefile:13: compile] Error 2
make[3]: Leaving directory '/home/ubuntu/Desktop/cm520/openwrt/target/linux'
time: target/linux/compile#4.99#35.43#26.08
make[2]: *** [target/Makefile:25: target/linux/compile] Error 2
make[2]: Leaving directory '/home/ubuntu/Desktop/cm520/openwrt'
make[1]: *** [target/Makefile:18: /home/ubuntu/Desktop/cm520/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/ubuntu/Desktop/cm520/openwrt'
make: *** [/home/ubuntu/Desktop/cm520/openwrt/include/toplevel.mk:236: world] Error 2

Lienol, 2020-11-03:
更新最新源码试试，应该能编译过，但我没有设备，不保证能用）

CMCCAdmin, 2020-11-04:
> 更新最新源码试试，应该能编译过，但我没有设备，不保证能用）
实体机依然报错，云编译过了但网口无法获取ip。

closed 2020-11-05 by CMCCAdmin