Lienol/openwrt #475: openwrt 设置nginx-ssl访问路由器管理界面

Dimitri2020007, 2020-11-09:
编译平台x86-64
分支19.07
已经在menuconfig luci-collections只选了luci-ssl-nginx。

已装了luci-acme

web server/proxies中也选择了nginx-ssl服务器与nginx-mod-luci-ssl支持

为什么只能用http而不是https访问路由器管理界面?

还是我又少添加啥软件了?

Lienol, 2020-11-11:
向官方反馈

closed 2020-11-11 by Lienol

Dimitri2020007, 2020-11-12:
问题已经向openwrt/package组反馈了。

VergilGao, 2020-11-13:
你能用https访问吗？我这里装了luci-ssl-nginx还是只能用http，用https访问就显示拒绝连接。
我手动装的acme.sh
在服务器上配置nginx都能跑，同样的配置在路由器上跑不了……

Dimitri2020007, 2020-11-14:
我现在碰到的就是这问题啊，我明明编译了luci-ssl-nginx可实际运行的时候只能是http（其他的取消了），https拒绝连接（lean有nginx-ssl-util所以可以直接https+nginx访问）

若使用luci-ssl-openssl（uhttpd的服务器）则可以正常访问https

我想用https的nginx真就这么难了吗？

VergilGao, 2020-11-14:
> 我现在碰到的就是这问题啊，我明明编译了luci-ssl-nginx可实际运行的时候只能是http（其他的取消了），https拒绝连接（lean有nginx-ssl-util所以可以直接https+nginx访问）
> 
> 若使用luci-ssl-openssl（uhttpd的服务器）则可以正常访问https
> 
> 我想用https的nginx真就这么难了吗？

 cat /var/log/nginx/error.log

你看一下报错信息

VergilGao, 2020-11-14:
> 向官方反馈

经过排查，uhttpd是默认选项，而且不可取消，即使安装了nginx-ssl，uhttpd仍然会是默认的web服务器，而nginx会尝试启动，因为端口被占用而永远无法启动。

Dimitri2020007, 2020-11-14:
这个我会注意下的

VergilGao, 2020-11-14:
> 这个我会注意下的，另外acme咋设置才能用在nginx上啊？

建议看 acme 官方文档，写的很详细。

Dimitri2020007, 2020-11-16:
取消webadmin无法登录ssh后台

Lienol, 2020-11-17:
这是因为https不能加载http资源，要么就把ttyd和netdata也换成https

Lienol, 2020-11-17:
要么就用nginx给反代一下

Dimitri2020007, 2020-11-18:
nginx反代倒是有可能，ttyd和netdata有https版本吗？我怎么找不到？还是要修改内核config？