Lienol/openwrt #485: configure: error: *** libpam missing

viagram, 2020-11-29:
2020-11-29T05:06:32.9773856Z checking whether OpenSSL's PRNG is internally seeded... checking for dlopen in -ldl... yes
2020-11-29T05:06:33.0214647Z checking for pam_set_item in -lpam... no
2020-11-29T05:06:33.0219474Z configure: error: *** libpam missing
2020-11-29T05:06:33.0669810Z Makefile:253: recipe for target '/home/runner/work/openwrt-build-test/openwrt-build-test/k3/build_dir/target-arm_cortex-a9_musl_eabi/openssh-with-pam/openssh-8.4p1/.configured_68b329da9893e34099c7d8ad5cb9c940' failed
2020-11-29T05:06:33.0673402Z make[3]: Leaving directory '/home/runner/work/openwrt-build-test/openwrt-build-test/k3/feeds/packages/net/openssh'
2020-11-29T05:06:33.0702670Z make[3]: *** [/home/runner/work/openwrt-build-test/openwrt-build-test/k3/build_dir/target-arm_cortex-a9_musl_eabi/openssh-with-pam/openssh-8.4p1/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
2020-11-29T05:06:33.0704324Z time: package/feeds/packages/openssh/with-pam/compile#14.65#7.08#19.33
2020-11-29T05:06:33.0726792Z make[2]: *** [package/feeds/packages/openssh/compile] Error 2
2020-11-29T05:06:33.0728038Z package/Makefile:112: recipe for target 'package/feeds/packages/openssh/compile' failed
2020-11-29T05:06:33.0729655Z make[1]: *** [/home/runner/work/openwrt-build-test/openwrt-build-test/k3/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.package_compile] Error 2
2020-11-29T05:06:33.0731336Z make[2]: Leaving directory '/home/runner/work/openwrt-build-test/openwrt-build-test/k3'
2020-11-29T05:06:33.0733043Z package/Makefile:108: recipe for target '/home/runner/work/openwrt-build-test/openwrt-build-test/k3/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.package_compile' failed
2020-11-29T05:06:33.0734862Z make[1]: Leaving directory '/home/runner/work/openwrt-build-test/openwrt-build-test/k3'
2020-11-29T05:06:33.0760772Z /home/runner/work/openwrt-build-test/openwrt-build-test/k3/include/toplevel.mk:240: recipe for target 'world' failed
2020-11-29T05:06:33.0764064Z make: *** [world] Error 2

viagram, 2020-11-29:
补充一下，我选择的是：openssh-server-pam
两天前的源码没问题

Lienol, 2020-11-30:
rm -rf bin tmp
./scripts/feeds clean
./scripts/feeds update -a
./scripts/feeds install -a
再重新试试

viagram, 2020-12-01:
It's OK!

closed 2020-12-01 by viagram