Lienol/openwrt #507: k3screenctrl编译错误

tporwx, 2020-12-22:
用的main分支，luci也换成了main,再加了ctcgfw那的两个包”luci-app-k3screenctrl“和“phicomm-k3screenctrl”，并且删除了lean目录下的“k3screenctrl”，编译出现错误，应该怎么解决呢？

> make[3]: Entering directory '/home/lxb/openwrt/package/ctcgfw/luci-app-k3screenctrl'
/home/lxb/openwrt/feeds/luci/luci.mk:306: *** Package/luci-i18n-k3screenctrl-zh-cn is missing the VERSION field.  Stop.
make[3]: Leaving directory '/home/lxb/openwrt/package/ctcgfw/luci-app-k3screenctrl'
time: package/ctcgfw/luci-app-k3screenctrl/compile#1.22#0.15#1.36
    ERROR: package/ctcgfw/luci-app-k3screenctrl failed to build.
make[2]: *** [package/Makefile:114: package/ctcgfw/luci-app-k3screenctrl/compile] Error 1
make[2]: Leaving directory '/home/lxb/openwrt'
make[1]: *** [package/Makefile:108: /home/lxb/openwrt/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lxb/openwrt'
make: *** [/home/lxb/openwrt/include/toplevel.mk:242: world] Error 2

trombonist852, 2021-02-18:
> 用的main分支，luci也换成了main,再加了ctcgfw那的两个包”luci-app-k3screenctrl“和“phicomm-k3screenctrl”，并且删除了lean目录下的“k3screenctrl”，编译出现错误，应该怎么解决呢？

打开/feeds/luci/luci.mk这个文件，找到第84行和93行，都改成echo "unknown"; \
再编译就可以了

Lienol, 2021-03-01:
fix

closed 2021-03-01 by Lienol