Lienol/openwrt #509: 不支持树莓派4b？

lexin8, 2020-12-30:
#其他的源码选项
Target System -> Broadcom BCM27xx
Subtarget -> BCM2711 boards (64 bit)
Target Profile -> Raspberry Pi 4B

#19.07选3B适配？
CONFIG_TARGET_brcm2708=y
CONFIG_TARGET_brcm2708_bcm2710=y
CONFIG_TARGET_brcm2708_bcm2710_DEVICE_rpi-3=y

Lienol, 2020-12-30:
19.07 没适配

wartw, 2021-05-18:
@Lienol 
Note, selecting 'zlib1g-dev' instead of 'libz-dev'
Note, selecting 'libpython3.5-minimal' for regex 'python3.5'
Note, selecting 'node-uglify' instead of 'uglifyjs'
Note, selecting 'git' instead of 'git-core'
Note, selecting 'upx-ucl' instead of 'upx'
Package gcc-multilib is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Unable to locate package lib32gcc1
E: Unable to locate package libc6-dev-i386
E: Package 'gcc-multilib' has no installation candidate
E: Unable to locate package g++-multilib
E: Couldn't find any package by regex 'g++-multilib'

wartw, 2021-05-18:
@lexin8 
sudo apt-get -y install build-essential asciidoc binutils bzip2 curl gawk gettext git libncurses5-dev zlib1g-dev patch  python2.7 unzip zlib1g-dev  libc6-dev subversion flex node-uglify gcc p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx-ucl libelf-dev autoconf automake libtool autopoint device-tree-compiler  antlr3 gperf g++
git clone -b 21.02 --single-branch https://github.com/Lienol/openwrt openwrt
cd openwrt
./scripts/feeds clean
./scripts/feeds update -a
./scripts/feeds install -a
make menuconfig
make -j4 download V=s
make -j4 V=s

closed 2021-07-05 by lexin8