Lienol/openwrt #520: docker安装后没有容器镜像等信息显示

starP-W, 2021-01-16:
安装后，luci左侧仅显示概况：
[image]
已安装的软件包如图所示：
[image]
系统日志报错信息如下：
```
Sun Jan 17 00:32:54 2021 user.notice dockerd-init: Drop traffic from eth1 to docker0
Sun Jan 17 00:32:54 2021 daemon.notice procd: /etc/init.d/dockerd: /etc/rc.common: /etc/init.d/dockerd: line 6: extra_command: not found
Sun Jan 17 00:32:54 2021 daemon.notice procd: /etc/init.d/dockerd: /etc/rc.common: /etc/init.d/dockerd: line 7: extra_command: not found
Sun Jan 17 00:32:56 2021 daemon.err modprobe: failed to find a module named aufs
Sun Jan 17 00:32:56 2021 daemon.err dockerd[4332]: time="2021-01-16T16:32:56.601860283Z" level=warning msg="failed to load plugin io.containerd.snapshotter.v1.devmapper" error="devmapper not configured"
Sun Jan 17 00:32:56 2021 daemon.err dockerd[4332]: time="2021-01-16T16:32:56.602065735Z" level=warning msg="could not use snapshotter devmapper in metadata plugin" error="devmapper not configured"
Sun Jan 17 00:32:56 2021 daemon.err dockerd[4332]: time="2021-01-16T16:32:56.639213135Z" level=warning msg="unable to modify root key limit, number of containers could be limited by this quota: open /proc/sys/kernel/keys/root_maxkeys: no such file or directory"
Sun Jan 17 00:32:56 2021 daemon.err dockerd[4332]: time="2021-01-16T16:32:56.652498401Z" level=error msg="failed to mount overlay: invalid argument" storage-driver=overlay2
Sun Jan 17 00:32:56 2021 daemon.err dockerd[4332]: time="2021-01-16T16:32:56.652538110Z" level=error msg="exec: \"fuse-overlayfs\": executable file not found in $PATH" storage-driver=fuse-overlayfs
Sun Jan 17 00:32:56 2021 daemon.err modprobe: failed to find a module named aufs
Sun Jan 17 00:32:56 2021 daemon.err dockerd[4332]: time="2021-01-16T16:32:56.654718785Z" level=error msg="AUFS was not found in /proc/filesystems" storage-driver=aufs
Sun Jan 17 00:32:56 2021 daemon.err dockerd[4332]: time="2021-01-16T16:32:56.655716607Z" level=error msg="failed to mount overlay: invalid argument" storage-driver=overlay
Sun Jan 17 00:32:56 2021 daemon.err dockerd[4332]: time="2021-01-16T16:32:56.655730496Z" level=error msg="Failed to built-in GetDriver graph devicemapper /opt/docker"
Sun Jan 17 00:32:56 2021 kern.err kernel: [  504.731476] overlayfs: filesystem on '/opt/docker/check-overlayfs-support437094600/upper' not supported as upperdir
Sun Jan 17 00:32:56 2021 kern.err kernel: [  504.734655] overlayfs: filesystem on '/opt/docker/check-overlayfs-support701962631/upper' not supported as upperdir
Sun Jan 17 00:32:58 2021 daemon.err dockerd[4332]: panic: cannot statfs cgroup root
Sun Jan 17 00:32:58 2021 daemon.err dockerd[4332]:
Sun Jan 17 00:32:58 2021 daemon.err dockerd[4332]: goroutine 1 [running]:
Sun Jan 17 00:32:58 2021 daemon.err dockerd[4332]: github.com/docker/docker/vendor/github.com/opencontainers/runc/libcontainer/cgroups.IsCgroup2UnifiedMode.func1()
Sun Jan 17 00:32:58 2021 daemon.err dockerd[4332]:  /path/of/compile/openwrt19/build_dir/target-x86_64_musl/dockerd-20.10.2/.go_work/build/src/github.com/docker/docker/vendor/github.com/opencontainers/runc/libcontainer/cgroups/utils.go:44 +0xc5
Sun Jan 17 00:32:58 2021 daemon.err dockerd[4332]: sync.(*Once).doSlow(0x39b3bf8, 0x28d7338)
Sun Jan 17 00:32:58 2021 daemon.err dockerd[4332]:  /path/of/compile/openwrt19/staging_dir/hostpkg/lib/go-cross/src/sync/once.go:66 +0xee
Sun Jan 17 00:32:58 2021 daemon.err dockerd[4332]: sync.(*Once).Do(...)
Sun Jan 17 00:32:58 2021 daemon.err dockerd[4332]:  /path/of/compile/openwrt19/staging_dir/hostpkg/lib/go-cross/src/sync/once.go:57
Sun Jan 17 00:32:58 2021 daemon.err dockerd[4332]: github.com/docker/docker/vendor/github.com/opencontainers/runc/libcontainer/cgroups.IsCgroup2UnifiedMode(0x0)
Sun Jan 17 00:32:58 2021 daemon.err dockerd[4332]:  /path/of/compile/openwrt19/build_dir/target-x86_64_musl/dockerd-20.10.2/.go_work/build/src/github.com/docker/docker/vendor/github.com/opencontainers/runc/libcontainer/cgroups/utils.go:41 +0x5a
Sun Jan 17 00:32:58 2021 daemon.err dockerd[4332]: github.com/docker/docker/vendor/github.com/opencontainers/runc/libcontainer/cgroups.GetCgroupMounts(0x203000, 0x203000, 0xc000b46de8, 0x4e5b51, 0xc000a86ea0, 0x0)
Sun Jan 17 00:32:58 2021 daemon.err dockerd[4332]:  /path/of/compile/openwrt19/build_dir/target-x86_64_musl/dockerd-20.10.2/.go_work/build/src/github.com/docker/docker/vendor/github.com/opencontainers/runc/libcontainer/cgroups/utils.go:62 +0x28
Sun Jan 17 00:32:58 2021 daemon.err dockerd[4332]: github.com/docker/docker/pkg/sysinfo.findCgroupMountpoints(0x2848900, 0xc000a84550, 0xc000b46f40)
Sun Jan 17 00:32:58 2021 daemon.err dockerd[4332]:  /path/of/compile/openwrt19/build_dir/target-x86_64_musl/dockerd-20.10.2/.go_work/build/src/github.com/docker/docker/pkg/sysinfo/sysinfo_linux.go:18 +0x3d
Sun Jan 17 00:32:58 2021 daemon.err dockerd[4332]: github.com/docker/docker/pkg/sysinfo.New(0xc0001b0600, 0x0, 0x0, 0x0, 0x2918fe0)
Sun Jan 17 00:32:58 2021 daemon.err dockerd[4332]:  /path/of/compile/openwrt19/build_dir/target-x86_64_musl/dockerd-20.10.2/.go_work/build/src/github.com/docker/docker/pkg/sysinfo/sysinfo_linux.go:67 +0xc5
Sun Jan 17 00:32:58 2021 daemon.err dockerd[4332]: github.com/docker/docker/daemon.(*Daemon).RawSysInfo(0xc00000c1e0, 0xc0001b0600, 0x0)
Sun Jan 17 00:32:58 2021 daemon.err dockerd[4332]:  /path/of/compile/openwrt19/build_dir/target-x86_64_musl/dockerd-20.10.2/.go_work/build/src/github.com/docker/docker/daemon/daemon_unix.go:1717 +0x9c
Sun Jan 17 00:32:58 2021 daemon.err dockerd[4332]: github.com/docker/docker/daemon.NewDaemon(0x297a260, 0xc0009e2bc0, 0xc0001b0680, 0xc000a200f0, 0x0, 0x0, 0x0)
Sun Jan 17 00:32:58 2021 daemon.err dockerd[4332]:  /path/of/compile/openwrt19/build_dir/target-x86_64_musl/dockerd-20.10.2/.go_work/build/src/github.com/docker/docker/daemon/daemon.go:1076 +0x244a
Sun Jan 17 00:32:58 2021 daemon.err dockerd[4332]: main.(*DaemonCli).start(0xc0009d15c0, 0xc00012fc80, 0x0, 0x0)
Sun Jan 17 00:32:58 2021 daemon.err dockerd[4332]:  /path/of/compile/openwrt19/build_dir/target-x86_64_musl/dockerd-20.10.2/.go_work/build/src/github.com/docker/docker/cmd/dockerd/daemon.go:195 +0x756
Sun Jan 17 00:32:58 2021 daemon.err dockerd[4332]: main.runDaemon(...)
Sun Jan 17 00:32:58 2021 daemon.err dockerd[4332]:  /path/of/compile/openwrt19/build_dir/target-x86_64_musl/dockerd-20.10.2/.go_work/build/src/github.com/docker/docker/cmd/dockerd/docker_unix.go:13
Sun Jan 17 00:32:58 2021 daemon.err dockerd[4332]: main.newDaemonCommand.func1(0xc0009aeb00, 0xc0009c9270, 0x0, 0x1, 0x0, 0x0)
Sun Jan 17 00:32:58 2021 daemon.err dockerd[4332]:  /path/of/compile/openwrt19/build_dir/target-x86_64_musl/dockerd-20.10.2/.go_work/build/src/github.com/docker/docker/cmd/dockerd/docker.go:34 +0x7d
Sun Jan 17 00:32:58 2021 daemon.err dockerd[4332]: github.com/docker/docker/vendor/github.com/spf13/cobra.(*Command).execute(0xc0009aeb00, 0xc000122030, 0x1, 0x1, 0xc0009aeb00, 0xc000122030)
Sun Jan 17 00:32:58 2021 daemon.err dockerd[4332]:  /path/of/compile/openwrt19/build_dir/target-x86_64_musl/dockerd-20.10.2/.go_work/build/src/github.com/docker/docker/vendor/github.com/spf13/cobra/command.go:850 +0x47c
Sun Jan 17 00:32:58 2021 daemon.err dockerd[4332]: github.com/docker/docker/vendor/github.com/spf13/cobra.(*Command).ExecuteC(0xc0009aeb00, 0x0, 0x0, 0x10)
Sun Jan 17 00:32:58 2021 daemon.err dockerd[4332]:  /path/of/compile/openwrt19/build_dir/target-x86_64_musl/dockerd-20.10.2/.go_work/build/src/github.com/docker/docker/vendor/github.com/spf13/cobra/command.go:958 +0x375
Sun Jan 17 00:32:58 2021 daemon.err dockerd[4332]: github.com/docker/docker/vendor/github.com/spf13/cobra.(*Command).Execute(...)
Sun Jan 17 00:32:58 2021 daemon.err dockerd[4332]:  /path/of/compile/openwrt19/build_dir/target-x86_64_musl/dockerd-20.10.2/.go_work/build/src/github.com/docker/docker/vendor/github.com/spf13/cobra/command.go:895
Sun Jan 17 00:32:58 2021 daemon.err dockerd[4332]: main.main()
Sun Jan 17 00:32:58 2021 daemon.err dockerd[4332]:  /path/of/compile/openwrt19/build_dir/target-x86_64_musl/dockerd-20.10.2/.go_work/build/src/github.com/docker/docker/cmd/dockerd/docker.go:97 +0x17d
```
想问一下要正常使用docker还需要进行什么操作或安装什么软件包？

weiweiqingshan, 2021-01-17:
在utilities选择dockerd

starP-W, 2021-01-17:
> 在utilities选择dockerd

dockerd已经安装了（见图2），而且如果没装的话图1里的概况→设置是只有dockerman一个tab的。

gww014, 2021-02-17:
同样的问题，不知道楼主解决了没有

starP-W, 2021-02-19:
> 同样的问题，不知道楼主解决了没有

解决是解决了，用最新的源码重新编译就好了，但是现在luci界面没法配置镜像和存储目录，实际上还是很不方便，也没什么用……

closed 2021-02-19 by starP-W