Lienol/openwrt #526: Adguardhome 保存在关机时备份工作目录文件报错，无法保存

MedyMa, 2021-02-18:
Adguardhome 保存’在关机时备份工作目录文件’时报错，无法保存
报错为: error!backup dir is a file
[image]

Jingye-Xia, 2021-02-19:
你为什么要‘在关机时备份工作目录文件’呢？

MedyMa, 2021-02-19:
> 你为什么要‘在关机时备份工作目录文件’呢？

有这选项我为啥不能选呢？不然这个功能就么啥用。。。

Jingye-Xia, 2021-02-19:
> > 你为什么要‘在关机时备份工作目录文件’呢？
> 
> 有这选项我为啥不能选呢？不然这个功能就么啥用。。。

你选这个选项是想实现什么功能呢？

MedyMa, 2021-02-19:
> > > 你为什么要‘在关机时备份工作目录文件’呢？
> > 
> > 
> > 有这选项我为啥不能选呢？不然这个功能就么啥用。。。
> 
> 你选这个选项是想实现什么功能呢？

就是关机备份一下配置

Jingye-Xia, 2021-02-19:
> > > > 你为什么要‘在关机时备份工作目录文件’呢？
> > > 
> > > 
> > > 有这选项我为啥不能选呢？不然这个功能就么啥用。。。
> > 
> > 
> > 你选这个选项是想实现什么功能呢？
> 
> 就是关机备份一下配置

关机配置不会被清除。。

Jingye-Xia, 2021-02-19:
> > > > 你为什么要‘在关机时备份工作目录文件’呢？
> > > 
> > > 
> > > 有这选项我为啥不能选呢？不然这个功能就么啥用。。。
> > 
> > 
> > 你选这个选项是想实现什么功能呢？
> 
> 就是关机备份一下配置

而且，luci-app-adguardhome有任何问题的话，你来错地方了，luci-app-adguardhome并不是Lienol写的，你应该去https://github.com/rufengsuixing/luci-app-adguardhome反馈。

MedyMa, 2021-02-19:
> > > > > 你为什么要‘在关机时备份工作目录文件’呢？
> > > > 
> > > > 
> > > > 有这选项我为啥不能选呢？不然这个功能就么啥用。。。
> > > 
> > > 
> > > 你选这个选项是想实现什么功能呢？
> > 
> > 
> > 就是关机备份一下配置
> 
> 关机配置不会被清除。。

那好吧，请原谅我是个强迫症患者