Lienol/openwrt #528: 编译qbittorrent出错，还望大神们不吝赐教，感激不尽！

trombonist852, 2021-02-18:
21.02编译树莓派4b **和x86，都提示找不到boost的版本** 但是在menuconfig里面选中boost之后 还是会跳错误，这个如何破？望大神帮忙看看 谢谢啦！

这是没选boost之前的错误：
checking for Qt5 qmake >= 5.9.0... /home/trb/21.02/staging_dir/toolchain-aarch64_cortex-a72_gcc-8.4.0_musl/bin/qmake
checking whether QtDBus should be enabled... no
checking for boostlib >= 1.40 (104000) includes in "/home/trb/21.02/staging_dir/target-aarch64_cortex-a72_musl/usr/include"... yes
checking for boostlib >= 1.40 (104000) lib path in "/home/trb/21.02/staging_dir/target-aarch64_cortex-a72_musl/usr/lib/aarch64-linux-gnu"... no
checking for boostlib >= 1.40 (104000) lib path in "/home/trb/21.02/staging_dir/target-aarch64_cortex-a72_musl/usr/lib64"... no
checking for boostlib >= 1.40 (104000) lib path in "/home/trb/21.02/staging_dir/target-aarch64_cortex-a72_musl/usr/lib"... yes
checking for boostlib >= 1.40 (104000)... yes
configure: Boost CXXFLAGS: "-I/home/trb/21.02/staging_dir/target-aarch64_cortex-a72_musl/usr/include"
configure: Boost LDFLAGS: "-L/home/trb/21.02/staging_dir/target-aarch64_cortex-a72_musl/usr/lib"
checking whether the Boost::System library is available... yes
configure: error: Could not find a version of the Boost::System library!
make[2]: *** [Makefile:67: /home/trb/21.02/build_dir/target-aarch64_cortex-a72_musl/qBittorrent-release-4.3.1/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
make[2]: Leaving directory '/home/trb/21.02/package/lean/qBittorrent'
time: package/lean/qBittorrent/compile#4.01#0.17#4.09
ERROR: package/lean/qBittorrent failed to build.
make[1]: *** [package/Makefile:114: package/lean/qBittorrent/compile] Error 1
make[1]: Leaving directory '/home/trb/21.02'
make: *** [/home/trb/21.02/include/toplevel.mk:230：package/qBittorrent/compile] 错误 2

lexin8, 2021-02-20:
编译报错的除非是文件冲突，不然还是云编译走一遍看能不能过来排除问题，去掉多余的插件，单独编译该插件，如果不能过就等作者修复，能过就删除本地环境重新编译。这是最简单的解决方式。

jh0072110, 2021-02-20:
main分支现在都不合适呢老哥就勇敢上21.02了都？出错不是很正常么……

closed 2021-04-08 by Lienol