Lienol/openwrt #541: Strange problem running Xray on OpenWRT

M33FT, 2021-03-05:
I asked elsewhere and no one could help me but here I ask maybe someone can help. . .

Until two months ago, I was using OpenWRT => Lean => Passwall => Xray + VLESS + XTLS on devices (RaspberryPi3B, RaspberryPi4B, WRT32X, WRT1900ACS) and everything worked fine.

After the government of the country where I live started the DNS Hijack, I could no longer connect to our OpenWRT devices
(I should also say that I can now easily use xray on clients like V2rayN or Qv2ray on PC To). But after a lot of searching, I realized that this is a DNS problem and I have to use programs like https-dns-proxy or smartdns that support DOH and DOT.
The interesting thing about this story is that now NaiveProxy, Trojan (not xray Trojan) SS, SSR work, but in general Xray (VLESS, VMESS) does not work on OpenWRT.

I checked xray with clients like Passwall, Bypass, Hello World, SSR+, Luci-app-xray and on Lean, Lienol, immortalWRT ROMs and none of them worked. And to say that all kinds of DNS settings like Xray DNS, Socks2DNS, pdnsd and. . . I checked but none of them worked.

And a very interesting thing is .... I think the old version of xray, think early versions, which I installed on my friend RaspberryPi4B device (Lean + Passwall + https-app-dns) is still working and no problem until It has not happened to him now (even the internet servers are both the same). Unfortunately, I no longer have the files that I compiled for my friend.

I even used the famous compiled versions but unfortunately xray (with different settings) did not work.
For example, I have included the RaspberryPi4B device Config file for compiling ImmortalWRT 18.06.

```
CONFIG_TARGET_bcm27xx=y
CONFIG_TARGET_bcm27xx_bcm2711=y
CONFIG_TARGET_bcm27xx_bcm2711_DEVICE_rpi-4=y
CONFIG_DOCKER_CGROUP_OPTIONS=y
CONFIG_DOCKER_NET_MACVLAN=y
CONFIG_DOCKER_STO_EXT4=y
CONFIG_KERNEL_CGROUP_DEVICE=y
CONFIG_KERNEL_CGROUP_FREEZER=y
CONFIG_KERNEL_CGROUP_NET_PRIO=y
CONFIG_KERNEL_EXT4_FS_POSIX_ACL=y
CONFIG_KERNEL_EXT4_FS_SECURITY=y
CONFIG_KERNEL_FS_POSIX_ACL=y
CONFIG_KERNEL_NET_CLS_CGROUP=y
CONFIG_LIBCURL_COOKIES=y
CONFIG_LIBCURL_CRYPTO_AUTH=y
CONFIG_LIBCURL_FILE=y
CONFIG_LIBCURL_FTP=y
CONFIG_LIBCURL_HTTP=y
CONFIG_LIBCURL_LDAP=y
CONFIG_LIBCURL_LDAPS=y
CONFIG_LIBCURL_NGHTTP2=y
CONFIG_LIBCURL_OPENSSL=y
CONFIG_LIBCURL_PROXY=y
CONFIG_LIBCURL_TFTP=y
CONFIG_LIBCURL_THREADED_RESOLVER=y
CONFIG_LIBCURL_TLS_SRP=y
CONFIG_LIBCURL_UNIX_SOCKETS=y
# CONFIG_PACKAGE_UnblockNeteaseMusic is not set
# CONFIG_PACKAGE_UnblockNeteaseMusic-Go is not set
# CONFIG_PACKAGE_adbyby is not set
CONFIG_PACKAGE_bash=y
CONFIG_PACKAGE_blkid=y
CONFIG_PACKAGE_boost=y
CONFIG_PACKAGE_boost-date_time=y
CONFIG_PACKAGE_boost-program_options=y
CONFIG_PACKAGE_boost-system=y
CONFIG_PACKAGE_brook=y
CONFIG_PACKAGE_btrfs-progs=y
CONFIG_PACKAGE_cgroupfs-mount=y
CONFIG_PACKAGE_chinadns-ng=y
CONFIG_PACKAGE_containerd=y
CONFIG_PACKAGE_coreutils-nohup=y
CONFIG_PACKAGE_curl=y
CONFIG_PACKAGE_docker=y
CONFIG_PACKAGE_docker-compose=y
CONFIG_PACKAGE_dockerd=y
CONFIG_PACKAGE_fdisk=y
CONFIG_PACKAGE_frpc=y
CONFIG_PACKAGE_haproxy=y
CONFIG_PACKAGE_https-dns-proxy=y
CONFIG_PACKAGE_ipt2socks=y
# CONFIG_PACKAGE_ipt2socks-alt is not set
CONFIG_PACKAGE_iptables-mod-extra=y
CONFIG_PACKAGE_kcptun-client=y
[... 171 lines cut ...]
CONFIG_PACKAGE_trojan=y
CONFIG_PACKAGE_trojan-go=y
CONFIG_PACKAGE_trojan-plus=y
CONFIG_PACKAGE_ttyd=y
CONFIG_PACKAGE_uclibcxx=y
CONFIG_PACKAGE_unzip=y
CONFIG_PACKAGE_v2ray-plugin=y
CONFIG_PACKAGE_xray-geodata=y
CONFIG_PACKAGE_xray-plugin=y
CONFIG_TARGET_ROOTFS_PARTSIZE=450
CONFIG_TROJAN_GO_COMPRESS_UPX=y
# CONFIG_UnblockNeteaseMusic_Go is not set
# CONFIG_UnblockNeteaseMusic_NodeJS is not set
CONFIG_boost-compile-visibility-hidden=y
CONFIG_boost-runtime-shared=y
CONFIG_boost-static-and-shared-libs=y
CONFIG_boost-variant-release=y
CONFIG_PACKAGE_etherwake=y
CONFIG_PACKAGE_wol=y
```

Has anyone had a problem similar to mine?
Can anyone help me?

[image]

jh0072110, 2021-03-05:
If ur local DNS can not check ur vless domain name, you can try to use ur server's ip address

M33FT, 2021-03-06:
I did that too, buddy, but unfortunately it didn't

M33FT, 2021-03-06:
that comes to your mind?

jh0072110, 2021-03-06:
Emmm maybe u can try to change ur local DNS first? I guess...
But if ur router can't connect ur server's ip, I think that's ur Communications Operator‘s problem, maybe change ur server's ip is best way to fix everything...

jh0072110, 2021-03-06:
PS: maybe u can open an issue in this place