Lienol/openwrt #546: 请问怎么在lienol大佬的代码中加入docker-ce进行编译？？？

caliban511, 2021-03-07:
如题。
我试着在用大佬的代码编译成的固件中安装dockerman，结果怎么搞都没法正常使用。
后来在网上查，说是要在编译的时候就要把支持docker的核心组件选上。
然后我试着在feeds里加入以下以行：
src-git dockerce https://github.com/docker/docker-ce;master
src-git libdocker https://github.com/lisaac/luci-lib-docker
src-git dockerman https://github.com/lisaac/luci-app-dockerman
然而docker-ce就是没有办法出现，编译出来的固件也仍然无法使用docker。
请教一下，正确的做法是怎样的？

caliban511, 2021-03-07:
不管怎么折腾，结果都如下图，手动装上docker-ce及所有相关依赖也是一样：
[image]

lexin8, 2021-03-08:
https://github.com/Lienol/openwrt/issues/539
同样问题，没有得到回复解决！该docker好像不用docker-ce改成
	+luci-compat \
	+luci-lib-docker \
	+dockerd \
	+ttyd
单移植docker-ce没用，1月25日就改了代码到现在一直这样！

WYC-2020, 2021-03-09:
有这么麻烦？自己编译加入编译luci-app-dockerman这个不就行了，这个自动回去找需要哪些依赖项，编译出来就可以用啊

caliban511, 2021-03-09:
> 有这么麻烦？自己编译加入编译luci-app-dockerman这个不就行了，这个自动回去找需要哪些依赖项，编译出来就可以用啊

正常应该是这样，但目前lienol这套代码不行，不信你可以试一下。

sbwml, 2021-03-25:
> 有这么麻烦？自己编译加入编译luci-app-dockerman这个不就行了，这个自动回去找需要哪些依赖项，编译出来就可以用啊

确实如此，勾选 luci-app-dockerman 就可以工作。
[image]

caliban511, 2021-03-25:
> 
> 
> > 有这么麻烦？自己编译加入编译luci-app-dockerman这个不就行了，这个自动回去找需要哪些依赖项，编译出来就可以用啊
> 
> 确实如此，勾选 luci-app-dockerman 就可以工作。
> [image]

你来迟了，前段时间大佬修复了这个问题。

lexin8, 2021-03-25:
> > 有这么麻烦？自己编译加入编译luci-app-dockerman这个不就行了，这个自动回去找需要哪些依赖项，编译出来就可以用啊
> 
> 确实如此，勾选 luci-app-dockerman 就可以工作。
> [image]

你们骗我！
[image]
[image]

caliban511, 2021-03-25:
> 
> 你们骗我！

呃……我上周编译的固件没这个问题了。。。

lexin8, 2021-03-25:
> > 你们骗我！
> 
> 呃……我上周编译的固件没这个问题了。。。

19.07有问题

yyetss, 2021-04-03:
19.07编译 了2次都是同样问题

Lienol, 2021-04-08:
已修复

closed 2021-04-08 by Lienol