Lienol/openwrt #556: AdGuard Home默认配置会无法启用

wto123pk, 2021-03-12:
AdGuard Home默认配置会无法启用，报“error!bin path is a dir”
默认：
执行文件路径：/usr/bin/AdGuardHome
修改为：
执行文件路径：/usr/bin/AdGuardHome/AdGuardHome
即可。
希望大大后期编译的时候能不能修改下。

wlmqwyf, 2021-03-15:
这是哪个分支？21.02？19.07？

wto123pk, 2021-03-15:
19.07的

wlmqwyf, 2021-03-16:
> 19.07
> [image]
> ![1](https://user-images.githubusercontent.com/80398261/111109528-e0055800-8595-11eb-9644-4651a87280e0.jpg

你是不是安装了俩AdGuardHome导致包冲突？

wto123pk, 2021-03-16:
> > 19.07 
> > ！[1]（https://user-images.githubusercontent.com/80398261/111109528-e0055800-8595-11eb-9644-4651a87280e0.jpg
> > [image][](https://user-images.githubusercontent.com/80398261/111109528-e0055800-8595-11eb-9644-4651a87280e0.jpg)
> 
> 你是不是安装了俩AdGuardHome导致包冲突？

这个倒没有，刷好固件后什么也没动过。默认就是这样。手动输一下就好。

closed 2021-04-08 by Lienol