Lienol/openwrt #569: [Main] X86_64使用测试5.10 kernel 编译报错

HanzChen, 2021-04-14:
请问5.10是否已经可以正常使用？我看commits里挺多5.10的？
当menuconfig里选择test kernel的时候有如下报错

SHELL= flock /home/ubuntu/openwrt-main/tmp/.root-copy.flock -c 'cp -fpR /home/ubuntu/openwrt-main/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kernel/. /home/ubuntu/openwrt-main/staging_dir/target-x86_64_musl/root-x86/'
touch /home/ubuntu/openwrt-main/staging_dir/target-x86_64_musl/root-x86/stamp/.kernel_installed
Packaged contents of /home/ubuntu/openwrt-main/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kernel into /home/ubuntu/openwrt-main/bin/targets/x86/64/packages/kernel_5.10.28-1-0b16f112384de68e4741988b921828bb_x86_64.ipk
NOTICE: module '/home/ubuntu/openwrt-main/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.28/drivers/scsi/scsi_mod.ko' is built-in.
NOTICE: module '/home/ubuntu/openwrt-main/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.28/drivers/scsi/sd_mod.ko' is built-in.
Packaged contents of /home/ubuntu/openwrt-main/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-scsi-core into /home/ubuntu/openwrt-main/bin/targets/x86/64/packages/kmod-scsi-core_5.10.28-1_x86_64.ipk
NOTICE: module '/home/ubuntu/openwrt-main/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.28/crypto/crypto_hash.ko' is built-in.
Packaged contents of /home/ubuntu/openwrt-main/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-crypto-hash into /home/ubuntu/openwrt-main/bin/targets/x86/64/packages/kmod-crypto-hash_5.10.28-1_x86_64.ipk
NOTICE: module '/home/ubuntu/openwrt-main/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.28/crypto/crypto_null.ko' is built-in.
Packaged contents of /home/ubuntu/openwrt-main/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-crypto-null into /home/ubuntu/openwrt-main/bin/targets/x86/64/packages/kmod-crypto-null_5.10.28-1_x86_64.ipk
NOTICE: module '/home/ubuntu/openwrt-main/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.28/crypto/aead.ko' is built-in.
NOTICE: module '/home/ubuntu/openwrt-main/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.28/crypto/geniv.ko' is built-in.
Packaged contents of /home/ubuntu/openwrt-main/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-crypto-aead into /home/ubuntu/openwrt-main/bin/targets/x86/64/packages/kmod-crypto-aead_5.10.28-1_x86_64.ipk
Packaged contents of /home/ubuntu/openwrt-main/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-crypto-arc4 into /home/ubuntu/openwrt-main/bin/targets/x86/64/packages/kmod-crypto-arc4_5.10.28-1_x86_64.ipk
NOTICE: module '/home/ubuntu/openwrt-main/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.28/crypto/crc32_generic.ko' is built-in.
Packaged contents of /home/ubuntu/openwrt-main/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-crypto-crc32 into /home/ubuntu/openwrt-main/bin/targets/x86/64/packages/kmod-crypto-crc32_5.10.28-1_x86_64.ipk
NOTICE: module '/home/ubuntu/openwrt-main/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.28/crypto/crc32c_generic.ko' is built-in.
Packaged contents of /home/ubuntu/openwrt-main/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-crypto-crc32c into /home/ubuntu/openwrt-main/bin/targets/x86/64/packages/kmod-crypto-crc32c_5.10.28-1_x86_64.ipk
Packaged contents of /home/ubuntu/openwrt-main/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-crypto-pcompress into /home/ubuntu/openwrt-main/bin/targets/x86/64/packages/kmod-crypto-pcompress_5.10.28-1_x86_64.ipk
NOTICE: module '/home/ubuntu/openwrt-main/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.28/crypto/cryptomgr.ko' is built-in.
Packaged contents of /home/ubuntu/openwrt-main/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-crypto-manager into /home/ubuntu/openwrt-main/bin/targets/x86/64/packages/kmod-crypto-manager_5.10.28-1_x86_64.ipk
NOTICE: module '/home/ubuntu/openwrt-main/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.28/crypto/ecb.ko' is built-in.
Packaged contents of /home/ubuntu/openwrt-main/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-crypto-ecb into /home/ubuntu/openwrt-main/bin/targets/x86/64/packages/kmod-crypto-ecb_5.10.28-1_x86_64.ipk
NOTICE: module '/home/ubuntu/openwrt-main/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.28/crypto/gf128mul.ko' is built-in.
Packaged contents of /home/ubuntu/openwrt-main/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-crypto-gf128 into /home/ubuntu/openwrt-main/bin/targets/x86/64/packages/kmod-crypto-gf128_5.10.28-1_x86_64.ipk
NOTICE: module '/home/ubuntu/openwrt-main/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.28/crypto/xts.ko' is built-in.
Packaged contents of /home/ubuntu/openwrt-main/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-crypto-xts into /home/ubuntu/openwrt-main/bin/targets/x86/64/packages/kmod-crypto-xts_5.10.28-1_x86_64.ipk
ERROR: module '/home/ubuntu/openwrt-main/build_dir/target-x86_64_musl/linux-x86_64/linux-5.10.28/crypto/anubis.ko' is missing.
make[3]: *** [modules/crypto.mk:739: /home/ubuntu/openwrt-main/bin/targets/x86/64/packages/kmod-crypto-misc_5.10.28-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/ubuntu/openwrt-main/package/kernel/linux'
time: package/kernel/linux/compile#2.06#0.20#2.07
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:114: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/ubuntu/openwrt-main'
make[1]: *** [package/Makefile:108: /home/ubuntu/openwrt-main/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ubuntu/openwrt-main'
make: *** [/home/ubuntu/openwrt-main/include/toplevel.mk:230: world] Error 2

jh0072110, 2021-04-14:
什么还可以从menuconfig里改kernel @。@

HanzChen, 2021-04-14:
> 什么还可以从menuconfig里改kernel @。@
> 又一次孤陋寡闻=。=

[image]

jh0072110, 2021-04-14:
> > 什么还可以从menuconfig里改kernel @。@
> > 又一次孤陋寡闻=。=
> 
> [image]

学习了学习了，我都直接改Makefile的=。=

HanzChen, 2021-04-14:
Kernel Modules ->  Cryptographic API modules->  []kmod-crypto-misc
取消选择这个kmod可以正常编译了

[image]

13810169, 2021-05-23:
我进去貌似没有这个测试版内核的选项啊........

HanzChen, 2021-05-23:
> 我进去貌似没有这个测试版内核的选项啊........

main分支才有

colinhd8, 2021-07-06:
666，碰到了同样的问题。但不知道取消这个会有何影响呢？

HanzChen, 2021-07-06:
> 666，碰到了同样的问题。但不知道取消这个会有何影响呢？

目前使用看来没什么影响。。我也不确定

colinhd8, 2021-07-06:
> > 666，碰到了同样的问题。但不知道取消这个会有何影响呢？
> 
> 目前使用看来没什么影响。。我也不确定

谢谢。好像5.4里面shortcfe，一开启此选项，就会编译报错。你有此问题吗？

closed 2021-07-18 by Lienol